moveit/moveit2 #3160: `MoveGroupInterface`'s `asyncExecute` corrupts memory, `execute` does not synchronize properly.

liarokapisv, 2024-12-12:
Hi there,

During my attempt to integrate `MoveGroupInterface` with [BehaviorTree](https://www.behaviortree.dev/) which I documented at https://github.com/moveit/moveit2/issues/3159, I looked at the [`execute`](https://github.com/moveit/moveit2/blob/abb440d1134ebab83ae3a87e46c45456247b2349/moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp#L792) method.
I noticed that [local variables](https://github.com/moveit/moveit2/blob/abb440d1134ebab83ae3a87e46c45456247b2349/moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp#L738-L740) are captured by the action callbacks *by reference.* 

Crucially, when the `async` variant is used, the function immediately returns, destroying the locals but the callbacks still reference and use them potentially leading to memory corruption.

Also, the `done` variable is not properly synchronized which technically affects the blocking version as well.

mikeferguson, 2024-12-12:
I think we can probably just merge these comments with your other ticket and update the description to basically be "async interface not useable" - a quick at the code leads me to believe that is probably the case.

liarokapisv, 2024-12-12:
I guess we could merge them. I figured these are technically separate issues, the `asyncExecute` method could very well just work without any state querying interface, which I guess is limiting but still. Fixing this issue by making these member variables and adding proper synchronization would of course make adding some state query functions trivial.

Main issue is that the other ticker is a feature request, how would you like to proceed?

robkwan, 2025-10-16:
Hello! I am new to this open-source contribution and I would like to ask if it is ok to try working on it as "good first issue" please.  I checked it my fix to my own forked branch only yet but I haven't create any pull request till I get ok to do so.  Thanks.

robkwan, 2025-11-05:
Hi @Prathmesh2931 yes, please take it over.  Thanks.