moveit/moveit2 #3162: RViz crashes when clicking "Reset" button while using the Planning Plugin

matt-shinfang, 2024-12-13:
### Description

I've encountered an issue where RViz crashes when clicking the "Reset" button while using the Planning Plugin. 
After investigating, it seems that the crash might be caused by  [this function](https://github.com/moveit/moveit2/blob/d13d835ebb66111fda80569547b729f43b2ecb49/moveit_ros/visualization/rviz_plugin_render_tools/src/mesh_shape.cpp#L138). RViz appears to attempt to clear a mesh that has already been removed.

My current quick and dirty workaround is to add a check for the existence of the mesh before attempting to remove it. Here is the modified clear() function:
```
void MeshShape::clear()
{
  if (entity_)
  {
    entity_->detachFromParent();
    const auto& mesh_name = entity_->getMesh()->getName(); 
    if(Ogre::MeshPtr mesh = Ogre::MeshManager::getSingleton().getByName(mesh_name)
    {
        Ogre::MeshManager::getSingleton().remove(mesh);
    }
    scene_manager_->destroyEntity(entity_);
    entity_ = nullptr;
  }
  manual_object_->clear();
  started_ = false;
}
```
Not sure if this is the best approach. Any suggestions, ideas, or feedback are welcome.

### ROS Distro

Humble

### OS and version

Ubuntu 22.04

### Source or binary build?

Source

### If source, which branch?

humble

### Which RMW are you using?

CycloneDDS

### Steps to Reproduce

1. Use docker image: `moveit/moveit2:main-humble-tutorial-source`
2. Run` ros2 launch moveit2_tutorials demo.launch.py`
3. Add a mesh into planning scene (I did this by using Rviz Scene Object tab)
4. Press reset

### Expected behavior

Reset Rviz.

### Actual behavior

Rviz crashed.

### Backtrace or Console output

 Console output:
```
[rviz2-4] [ERROR] [1733903296.600162075] [rviz2]: RuntimeAssertionException: attempting to remove unknown resource: ConvertedMeshShape@70 in group General in remove at /root/ws_moveit/build/rviz_ogre_vendor/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreResourceManager.cpp (line 301)
[rviz2-4] terminate called after throwing an instance of 'Ogre::RuntimeAssertionException'
[rviz2-4]   what():  RuntimeAssertionException: attempting to remove unknown resource: ConvertedMeshShape@70 in group General in remove at /root/ws_moveit/build/rviz_ogre_vendor/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreResourceManager.cpp (line 301)
```
Backtrace:
https://gist.github.com/matt-shinfang/2a58f8749f58f356ee403442dcb5dd2c

sea-bass, 2024-12-13:
That seems sensical. I've fixed similar issues with rviz plugins this way in the past.

If you can put up a PR to the main branch, we can get this in and backport it.

scmwang, 2024-12-16:
Sounds good! My PR :  https://github.com/moveit/moveit2/pull/3164

closed 2024-12-16 by sjahr