moveit/moveit2 #3165: Generated launch files from setup assistant not working

PietroValerio, 2024-12-16:
### Description

I tried to run the move group launch file generated from the setup assistant, and i get only an error instead of working.
```bash
ros2 launch ur_cell_moveit_config move_group.launch.py --debug
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[INFO] [launch]: All log files can be found below /home/valerio/.ros/log/2024-12-16-12-28-55-613277-Hades-6998
[INFO] [launch]: Default logging verbosity is set to DEBUG
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7d169b32c4a0>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7d169b32c4a0>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x7d169a1c2180>'
Executing <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/jazzy/lib/python3.12/site-packages/launch/launch_service.py:228> exception=TypeError("Expected 'value' to be one of [<class 'float'>, <class 'int'>, <class 'str'>, <class 'bool'>, <class 'bytes'>], but got '()' of type '<class 'tuple'>'") created at /opt/ros/jazzy/lib/python3.12/site-packages/launch/launch_service.py:319> took 0.102 seconds
[DEBUG] [launch]: An exception was raised in an async action/event
[DEBUG] [launch]: Traceback (most recent call last):
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch/launch_service.py", line 343, in run_async
    raise exception_to_raise
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch/launch_service.py", line 230, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch/launch_service.py", line 250, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  [Previous line repeated 1 more time]
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
                   ^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch/action.py", line 109, in visit
    return self.execute(context)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch_ros/actions/node.py", line 490, in execute
    self._perform_substitutions(context)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch_ros/actions/node.py", line 445, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch_ros/utilities/evaluate_parameters.py", line 169, in evaluate_parameters
    output_params.append(evaluate_parameter_dict(context, param))
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch_ros/utilities/evaluate_parameters.py", line 141, in evaluate_parameter_dict
    ensure_argument_type(value, (float, int, str, bool, bytes), 'value')
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch/utilities/ensure_argument_type_impl.py", line 73, in ensure_argument_type
    raise TypeError(error_msg_template.format(
TypeError: Expected 'value' to be one of [<class 'float'>, <class 'int'>, <class 'str'>, <class 'bool'>, <class 'bytes'>], but got '()' of type '<class 'tuple'>'

[ERROR] [launch]: Caught exception in launch (see debug for traceback): Expected 'value' to be one of [<class 'float'>, <class 'int'>, <class 'str'>, <class 'bool'>, <class 'bytes'>], but got '()' of type '<class 'tuple'>'
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.Shutdown'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7d169a8c9430>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7d169a8c9430>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7d169a02a690>'
```

### ROS Distro

Jazzy

### OS and version

Ubuntu 24.04

### Source or binary build?

Source

### If source, which branch?

main

### Which RMW are you using?

None

### Steps to Reproduce

run the generated launch file

### Expected behavior

should work

### Actual behavior

Error as shown in description

sea-bass, 2024-12-16:
What if you remove the `--debug`? I think that argument doesn't quite work like that for launch files.

PietroValerio, 2024-12-16:
> What if you remove the `--debug`? I think that argument doesn't quite work like that for launch files.

if i remove that argument I get only this error
`TypeError: Expected 'value' to be one of [<class 'float'>, <class 'int'>, <class 'str'>, <class 'bool'>, <class 'bytes'>], but got '()' of type '<class 'tuple'>'

[ERROR] [launch]: Caught exception in launch (see debug for traceback): Expected 'value' to be one of [<class 'float'>, <class 'int'>, <class 'str'>, <class 'bool'>, <class 'bytes'>], but got '()' of type '<class 'tuple'>'`

sea-bass, 2024-12-16:
Hmm... I might start commenting out different components in the launch file to see where the bad parameter definition is, then.

Quite likely there's a bug, but we'll need to find exactly where. Thanks!

mikeferguson, 2024-12-16:
From what I've seen, the two areas that are kinda brittle with regards the to the `move_group` launch file are the `moveit_controllers` and `sensors_3d` configurations - if you can post the related generated YAML files, we might notice something that is off

PietroValerio, 2024-12-16:
I don't use 3d sensors, so no config file,the moveit_controllers yaml is
```yaml
moveit_controller_manager: moveit_simple_controller_manager/MoveItSimpleControllerManager

moveit_simple_controller_manager:
  controller_names:
    - scaled_joint_trajectory_controller
    - ur_hand_controller

  scaled_joint_trajectory_controller:
    type: FollowJointTrajectory
    joints:
      - ur10e_shoulder_pan_joint
      - ur10e_shoulder_lift_joint
      - ur10e_elbow_joint
      - ur10e_wrist_1_joint
      - ur10e_wrist_2_joint
      - ur10e_wrist_3_joint
    action_ns: follow_joint_trajectory
    default: true
  ur_hand_controller:
    type: GripperCommand
    joints:
      - gripper_finger_joint
    action_ns: gripper_cmd
    default: true
```
If you need more info let me know!

sea-bass, 2024-12-17:
I would think the motion planning ones, like `ompl_planning.yaml` are also susceptible to having outdated generated structure. What do those look like?

sea-bass, 2024-12-20:
So it seems https://github.com/moveit/moveit2/issues/2837 highlights the same error, but points to an empty list of hand controller joints.

Do you by any chance also have another copy of this file with empty values? Or did your file start that way and maybe haven't rebuilt the package since?