moveit/moveit2 #3176: Stuck building moveit on humble for over 2.5 hrs

ronakbhanushali, 2024-12-21:
### Description

[image]

Tried building from source. All packages built except for moveit2_tutorials. Stuck at 92% since 2 hrs 31 mins now. 

### ROS Distro

Humble

### OS and version

Ubuntu 22.04.5

### Source or binary build?

Source

### Which RMW are you using?

None

### Steps to Reproduce

Followed tutorial and ran

`colcon build --mixin release`

### Expected behavior

Should build the packages

### Actual behavior

Stuck at moveit2_tutorials for 2 hr 31 mins

[Processing: moveit2_tutorials]                                                 
[Processing: moveit2_tutorials]                                                 
[Processing: moveit2_tutorials]                                                 
[Processing: moveit2_tutorials]                                                 
[Processing: moveit2_tutorials]                                                 
[Processing: moveit2_tutorials]                                                 
[Processing: moveit2_tutorials]                                                 
[Processing: moveit2_tutorials]                                                 
[Processing: moveit2_tutorials]                                                 
Aborted  <<< moveit2_tutorials [2h 31min 33s]

sea-bass, 2024-12-21:
Super strange! If you try to increase the verbosity level by adding `--event_handlers console direct+`, what do you see?

ronakbhanushali, 2024-12-21:
Tried running that but got an error

```
colcon build --mixin release --event_handlers console direct+
usage: colcon [-h] [--log-base LOG_BASE] [--log-level LOG_LEVEL]
              {build,extension-points,extensions,graph,info,list,metadata,mixin,test,test-result,version-check} ...
colcon: error: Mixin '--event_handlers' is not available for 'build'
```

Instead ran this
`colcon build --mixin release --event-handlers console_direct+ status+`

This is my output

[image]

It always stays stuck at 92% or 100%

ronakbhanushali, 2024-12-21:
For context, this is my system
[image]

AndyZe, 2024-12-21:
Possibly you're low on RAM? I think 32 GB should be enough, but... Try building with just a few threads with `colcon build --parallel-workers 2` or similar

ronakbhanushali, 2024-12-21:
Yeah, I already tried with 2 workers but it stays stuck. My system is not frozen, but the build just never finishes

ronakbhanushali, 2024-12-21:
[image]

Resource usage while building for reference

sea-bass, 2024-12-22:
I wonder if it's possibly related to X11 related issues as described in this issue:
https://github.com/colcon/colcon-ros/issues/122

This describes an issue with the popup window that can come up when build is completed, which may be failing to come up?

closed 2025-03-24 by github-actions[bot]