moveit/moveit2 #3177: I do not understand param "Min.collisions for 'always'-colliding pairs" in moveit setup assistant !

marcuswuh, 2024-12-24:
### Is your feature request related to a problem? Please describe.

[image]

Please help me.

### Describe the solution you'd like

I guess , this param is an threshold for 'always'-colliding pairs. but what exactly it is? an threshold for collision times or volume?

mikeferguson, 2024-12-24:
The general idea here is:

 * We sample a number of random poses of the robot (in your case, based on the slider, it looks like 10000 poses will be sampled)
 * For each sampled pose, we determine what links are in collision
 * After 10000 sampled poses, if >95% of the poses result in collision of the two links, they are considered to be "always colliding" - in your case, if in >9500 of the 10000 sampled poses.

mikeferguson, 2024-12-24:
It looks like the text in the setup assistant is being cut off for some reason (is the screen too small?) - here's what is in the code:

```
auto header = new HeaderWidget(
      "Optimize Self-Collision Checking",
      "This searches for pairs of robot links that can safely be disabled from collision checking, decreasing motion "
      "planning time. These pairs are disabled when they are always in collision, never in collision, in collision in "
      "the robot's default position, or when the links are adjacent to each other on the kinematic chain. Sampling "
      "density specifies how many random robot positions to check for self collision.",
      this);
```