moveit/moveit2 #3178: Problem with Setup Assistant 2.0

ItsTomSav, 2024-12-24:
### Description

I want to create a new robot configuration. I already have a .urdf.xacro file (I also tried with a .urdf file) ​​that loads in the Setup Assistant. When I get to the pose configuration in "Robot Pose" whatever slider I touch with the mouse, the Setup Assistant crashes.
I thought it was a problem with how my .urdf file was written, so I tried with the Panda .urdf file. I got the same problem.

### ROS Distro

Humble

### OS and version

Ubuntu 22.04

### Source or binary build?

Source

### If source, which branch?

Humble

### Which RMW are you using?

FastRTPS

### Steps to Reproduce

`ros2 launch moveit_setup_assistant setup_assistant.launch.py`

1. "Create new MoveIt Configuration Package"-> Select file.urdf -> Click on "Load FIles";
2. "Self-Collisions" Panel -> "Generate Collision Matrix";
3. Create Virtual Joint;
4. Create Planning Group (as it was done in the tutorial);
5. "Define Robot Pose" Panel -> "Add Pose" -> Click on any slider -> **_CRASHES_**

### Expected behavior

The Setup Assistant should not crash and should be able to create a new configuration file.

### Actual behavior

Crashes in "Robot Pose" Panel

### Backtrace or Console output

```
[moveit_setup_assistant-1] moveit_setup_assistant: /home/user/lab_ROS2/ws_moveit2/src/moveit2/moveit_core/robot_state/include/moveit/robot_state/robot_state.h:1397: const Isometry3d& moveit::core::RobotState::getCollisionBodyTransform(const moveit::core::LinkModel*, std::size_t) const: Assertion `checkCollisionTransforms()' failed.
[ERROR] [moveit_setup_assistant-1]: process has died [pid 1571, exit code -6, cmd '/home/user/lab_ROS2/ws_moveit2/install/moveit_setup_assistant/lib/moveit_setup_assistant/moveit_setup_assistant --ros-args'].
```

FrankSav80, 2024-12-24:
Sir I have the same problem

mikeferguson, 2024-12-24:
@ItsTomSav - which branch are you building on humble - `humble` or `main`?