moveit/moveit2 #3180: import Python pkg

HoryeongJung, 2024-12-26:
### Description

Here’s a concise explanation in English that you can post as a GitHub issue:

Problem: ImportError with _moveit_roscpp_initializer when using moveit_commander
Description
I'm working on MoveIt 2 in ROS 2 Humble and encountering an error while trying to use the Python moveit_commander module. After successfully building MoveIt 2 from source, I get the following error when importing moveit_commander:

hl9934@meric:~/ws_moveit/src/moveit2/moveit_ros/planning_interface$ python3 
Python 3.10.12 (main, Nov  6 2024, 20:22:13) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import moveit_commander
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/hl9934/ws_moveit/src/moveit2/moveit_commander/src/moveit_commander/__init__.py", line 2, in <module>
    from .roscpp_initializer import *
  File "/home/hl9934/ws_moveit/src/moveit2/moveit_commander/src/moveit_commander/roscpp_initializer.py", line 35, in <module>
    from moveit_ros_planning_interface import _moveit_roscpp_initializer
ImportError: cannot import name '_moveit_roscpp_initializer' from 'moveit_ros_planning_interface' (/home/hl9934/ws_moveit/src/moveit2/moveit_ros/planning_interface/python/moveit_ros_planning_interface/__init__.py)

I want to use moveit pkg as python not cpp. But when i import the moveit_commander, upper error always occurs. So i ask gpt, 

### ROS Distro

Humble

### OS and version

Ubuntu 22.04

### Source or binary build?

Source

### If source, which branch?

humble

### Which RMW are you using?

CycloneDDS

### Steps to Reproduce

.

### Expected behavior

.

### Actual behavior

.

sea-bass, 2024-12-26:
MoveIt Commander was not ported to ROS 2. It is misleading that this folder was not deleted in the Humble branch, but it is gone in later versions.

HoryeongJung, 2024-12-27:
Thanks! Then can u give me some python example using moveit source or let me know some way that use moveit resource to operate ur3 manipulator in real and gazebo?

sea-bass, 2024-12-28:
I would recommend looking at the Humble tutorials:
https://moveit.picknik.ai/humble/index.html

If you want to the use the MoveIt 2 Python bindings specifically, these are only available on newer versions, although you can also build the main branch of MoveIt from source on Humble.
https://moveit.picknik.ai/main/index.html

mikeferguson, 2024-12-28:
> It is misleading that this folder was not deleted in the Humble branch, but it is gone in later versions.

Should we remove it from the humble branch if it is unported?

sea-bass, 2024-12-28:
> Should we remove it from the humble branch if it is unported?

I think yes. Likely a lot of these deletion PRs exist already and just need backports.