moveit/moveit2 #3188: Pilz integration tests have started failing sporadically on Rolling due to exit code -11

sea-bass, 2025-01-02:
### Description

There have lately been pretty frequent (~50% or so) flakes with some exit codes in Pilz integration tests lately:
https://github.com/moveit/moveit2/actions/runs/12589290424/job/35088819135?pr=3187#step:11:8621

This seems to be specifically happening in the rolling + ikfast + clang-tidy CI job, but it's usually cleared away after 1 or 2 retries.

Creating this issue to keep track.

### ROS Distro

Rolling

### OS and version

Ubuntu 24.04

### Source or binary build?

Source

### Which RMW are you using?

None

### Steps to Reproduce

Run CI

### Expected behavior

CI is green

### Actual behavior

CI is not always green

GayarUsmanov, 2025-01-08:
Is there a way to reproduce it locally on my machine?

sea-bass, 2025-01-12:
> Is there a way to reproduce the issue locally on my machine?

In these integration tests I often rely on the `stress` command to reproduce locally: https://www.geeksforgeeks.org/linux-stress-command-with-examples/

However, issues pertaining to exit codes can be tricky and sometimes can't be reproduced. Worth a try, though.

GayarUsmanov, 2025-01-12:
Sorry, I should've been clearer. I meant to ask how do you run the integration test on local machine? 
Once I know how to do that I'll run it under a memory checker, e.g. Valgrind.

sea-bass, 2025-01-12:
Aha. You can do something like this to the invocation of `colcon test`

https://stackoverflow.com/a/77029260

MarcoMagriDev, 2025-02-12:
At the current stage I am not able to reproduce this issue locally.

---

For those of you are interested in reproducing this locally i suggest to use this command:
```bash
colcon build --cmake-clean-cache --mixin debug --packages-select pilz_industrial_motion_planner pilz_industrial_motion_planner_testutils && \
source install/setup.bash && \
colcon test --packages-select pilz_industrial_motion_planner --ctest-args --output-on-failure --interactive-debugger gdb
```

then using `colcon test-result --verbose` you can see the test outcome with gdb prints.

closed 2025-05-20 by github-actions[bot]