moveit/moveit2 #3201: How to add custom planner in moveit2

LiangXin-gugu, 2025-01-06:
Hello moveit community, I want to add a custom motion planner in moveit2.
My environment:

 1. Ubuntu22.04;
 2. ros2-humble;
 3. moveit2(from source).

I decide to take the first approach from [this](https://answers.ros.org/question/298345/different-ways-to-implement-custom-planners-with-moveit/): Install OMPL from source and add custom planner to OMPL, and then change the corresponding MoveIt OMPL package to recognize this new planner. 

I follow the steps as bellow (refer to [1](https://answers.ros.org/question/296238/custom-state-sampler-in-moveit/),[2](https://web.archive.org/web/20141025194602/http://moveit.ros.org/wiki/OMPL/Add_New_Planner)): 

 1. **Delete the OMPL binaries** using `sudo apt remove ros-humble-ompl`.
 2. **Download OMPL source code** using `git clone https://github.com/ompl/ompl.git` in custom dir "*~/gugu_workspace/ompl_ws/src/*".
 3. **Modify OMPL source code**. I just copy a RRT algorithm planner(copy *RRT.h* and *RRT.cpp* and change the name).
 4. **Change OMPL install target folder**, which involves changing both *.so binary* and *.h* files install path.
    - "*~/gugu_workspace/ompl_ws/src/ompl/src/ompl/CMakeList.txt*". On line 196, I change `DESTINATION ${CMAKE_INSTALL_LIBDIR}` to `DESTINATION "/opt/ros/humble/lib/x86_64-linux-gnu"`, which is the path where the OMPL *.so* binaries are original installed. 
    - "*/home/gugu/gugu_workspace/ompl_ws/src/ompl/src/CMakeLists.txt*". On line 14, I change `DESTINATION "${CMAKE_INSTALL_INCLUDEDIR}/ompl"` to `DESTINATION "/opt/ros/humble/include/ompl-1.6/ompl"`, which is the path where the OMPL *.h* headers are original installed. 
 5. **Install OMPL from source code** followed the installation tutorial in [this](https://ompl.kavrakilab.org/core/installation.html). 
 6. **Modify MoveIt source code**. Modify *planning_context_manager.cpp*: Include the header file that defines your planner at the top and register my planner in the registerDefaultPlanners() function as the existing ones are.
 7. **Rebuild MoveIt from source**. `cd ~/ws_moveit2` and `colcon build`.

After completing the above steps and compiling successfully, I try to test the planner through “MotionPlanning" Plugin in Rviz.
[![“MotionPlanning" Plugin][1]][1]
But once I set the goal state and click plan, the move_group node collapse. The messeages in terminal are as follows:

```
[INFO] [launch]: All log files can be found below /home/gugu/.ros/log/2024-11-17-08-49-36-152595-GuoGuo-284881
[INFO] [launch]: Default logging verbosity is set to INFO
WARNING:root:Cannot infer SRDF from `/home/gugu/gugu_workspace/simplified_ur10e_ws/install/simplified_ur10e_moveit_config_manual_setup/share/simplified_ur10e_moveit_config_manual_setup`. -- using config/ur10e.srdf
[INFO] [move_group-1]: process started with pid [284882]
[INFO] [rviz2-2]: process started with pid [284884]
[rviz2-2] QStandardPaths: wrong permissions on runtime directory /run/user/1000/, 0755 instead of 0700
[move_group-1] [INFO] [1731804576.501583939] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-1] [INFO] [1731804576.501684159] [moveit_robot_model.robot_model]: Loading robot model 'ur10e'...
[move_group-1] [INFO] [1731804576.501694045] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-1] [INFO] [1731804576.533358618] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-1] [INFO] [1731804576.533596528] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
[move_group-1] [INFO] [1731804576.534706727] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[move_group-1] [INFO] [1731804576.535451315] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-1] [INFO] [1731804576.535474916] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-1] [INFO] [1731804576.536201910] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-1] [INFO] [1731804576.536237015] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-1] [INFO] [1731804576.536724227] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-1] [INFO] [1731804576.537168789] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-1] [WARN] [1731804576.538114288] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-1] [ERROR] [1731804576.538149741] [moveit.ros.occupancy_map_monitor.middleware_handle]: No 3D sensor plugin(s) defined for octomap updates
[move_group-1] [INFO] [1731804576.539581807] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'ompl'
[move_group-1] [INFO] [1731804576.567914082] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[move_group-1] [INFO] [1731804576.574660182] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.path_tolerance' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1731804576.574736800] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.resample_dt' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1731804576.574743525] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.min_angle_change' was not set. Using default value: 0.001000
[move_group-1] [INFO] [1731804576.574778794] [moveit_ros.fix_workspace_bounds]: Param 'ompl.default_workspace_bounds' was not set. Using default value: 10.000000
[move_group-1] [INFO] [1731804576.574794556] [moveit_ros.fix_start_state_bounds]: Param 'ompl.start_state_max_bounds_error' was set to 0.100000
[move_group-1] [INFO] [1731804576.574800001] [moveit_ros.fix_start_state_bounds]: Param 'ompl.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-1] [INFO] [1731804576.574810266] [moveit_ros.fix_start_state_collision]: Param 'ompl.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-1] [INFO] [1731804576.574815292] [moveit_ros.fix_start_state_collision]: Param 'ompl.jiggle_fraction' was not set. Using default value: 0.020000
[move_group-1] [INFO] [1731804576.574834369] [moveit_ros.fix_start_state_collision]: Param 'ompl.max_sampling_attempts' was not set. Using default value: 100
[move_group-1] [INFO] [1731804576.574844952] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Add Time Optimal Parameterization'
[move_group-1] [INFO] [1731804576.574851646] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Resolve constraint frames to robot links'
[move_group-1] [INFO] [1731804576.574854645] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Workspace Bounds'
[move_group-1] [INFO] [1731804576.574857674] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Bounds'
[move_group-1] [INFO] [1731804576.574860560] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State In Collision'
[move_group-1] [INFO] [1731804576.574863365] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Path Constraints'
[move_group-1] [INFO] [1731804576.579468887] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'pilz_industrial_motion_planner'
[move_group-1] [INFO] [1731804576.591777385] [moveit.pilz_industrial_motion_planner.joint_limits_aggregator]: Reading limits from namespace robot_description_planning
[move_group-1] [INFO] [1731804576.597728437] [moveit.pilz_industrial_motion_planner]: Available plugins: pilz_industrial_motion_planner/PlanningContextLoaderCIRC pilz_industrial_motion_planner/PlanningContextLoaderLIN pilz_industrial_motion_planner/PlanningContextLoaderPTP 
[move_group-1] [INFO] [1731804576.597790992] [moveit.pilz_industrial_motion_planner]: About to load: pilz_industrial_motion_planner/PlanningContextLoaderCIRC
[move_group-1] [INFO] [1731804576.600146448] [moveit.pilz_industrial_motion_planner]: Registered Algorithm [CIRC]
[... cut ...]
-- Installing: /usr/local/lib/libompl.so.1.6.0
-- Installing: /usr/local/lib/libompl.so.17
-- Set runtime path of "/usr/local/lib/libompl.so.1.6.0" to "/usr/local/lib"
-- Installing: /usr/local/lib/libompl.so
```
Is it possible that the problem is caused by incorrect run time path settings?

My questions are:

 1. How to add custom motion planner in OMPL and then called by moveit2?
 2. Are there better ways to add custom motion planner in moveit2?

Any input is appreciated.

  [1]: https://i.sstatic.net/65o3xGUB.png

cairui-hub, 2025-02-13:
所以问题解决了么？我最近也需要解决这个问题，目前只知道ompl官网里有new planner.cpp模板文件，但具体这个文件怎么用没找到说明。

LiangXin-gugu, 2025-03-12:
> 所以问题解决了么？我最近也需要解决这个问题，目前只知道ompl官网里有new planner.cpp模板文件，但具体这个文件怎么用没找到说明。

解决了，我在这个链接上回答了。[link](https://robotics.stackexchange.com/questions/113731/how-to-add-custom-planner-in-moveit2/113754#113754)

LiangXin-gugu, 2025-03-12:
> I give my answer in this [link](https://robotics.stackexchange.com/questions/113731/how-to-add-custom-planner-in-moveit2/113754#113754), which works for me. Hope it will be helpful to someone :) ！

closed 2025-03-12 by LiangXin-gugu