moveit/moveit2 #3215: Servoing doesn't follow twist command correctly

PietroValerio, 2025-01-09:
### Description

I'm trying to do servoing, using the c++ api, all i've tried to do is to setting a target twist defined in the planning frame, in my case the robot frame, to move the end effector on the z-axis. As you can see in the video below this does not happen. I've taken the tutorial file used on franka emika panda and modified for working with ur10e.

https://github.com/user-attachments/assets/b772ccfd-968c-4403-b560-0a4604a94e9d

### ROS Distro

Jazzy

### OS and version

Ubuntu 24.04

### Source or binary build?

Source

### If source, which branch?

main commit abb440d1134ebab83ae3a87e46c45456247b2349

### Which RMW are you using?

None

### Steps to Reproduce

You need to have ros2 driver for ur robot provided by universal robots
Here is my modified .cpp file for servoing using the cpp api, i changed the type of command that the servo node outputs, because the /forward_position_controller/commands accepts only std_msgs::msg::Float64MultiArray
```cpp
#include <chrono>
#include <moveit_servo/servo.hpp>
#include <moveit_servo/utils/common.hpp>
#include <rclcpp/rclcpp.hpp>
#include <tf2_geometry_msgs/tf2_geometry_msgs.hpp>
#include <tf2_ros/transform_listener.h>
#include <moveit/utils/logger.hpp>

using namespace moveit_servo;

int main(int argc, char* argv[])
{
  rclcpp::init(argc, argv);

  const rclcpp::Node::SharedPtr demo_node = std::make_shared<rclcpp::Node>("moveit_servo_demo");
  moveit::setNodeLoggerName(demo_node->get_name());

 
  const std::string param_namespace = "moveit_servo";
  const std::shared_ptr<const servo::ParamListener> servo_param_listener =
      std::make_shared<const servo::ParamListener>(demo_node, param_namespace);
  const servo::Params servo_params = servo_param_listener->get_params();

//THIS LINE WAS MODIFIED
  rclcpp::Publisher<std_msgs::msg::Float64MultiArray>::SharedPtr trajectory_outgoing_cmd_pub =
      demo_node->create_publisher<std_msgs::msg::Float64MultiArray>(servo_params.command_out_topic,
                                                                         rclcpp::SystemDefaultsQoS());

 
  const planning_scene_monitor::PlanningSceneMonitorPtr planning_scene_monitor =
      createPlanningSceneMonitor(demo_node, servo_params);
  Servo servo = Servo(demo_node, servo_param_listener, planning_scene_monitor);

  std::this_thread::sleep_for(std::chrono::seconds(10));

  auto robot_state = planning_scene_monitor->getStateMonitor()->getCurrentState();
  const moveit::core::JointModelGroup* joint_model_group =
      robot_state->getJointModelGroup(servo_params.move_group_name);

[... 112 lines cut ...]

## Collision checking for the entire robot body
check_collisions: true # Check collisions?
collision_check_rate: 5.0 # [Hz] Collision-checking can easily bog down a CPU if done too often.
# Two collision check algorithms are available: "threshold_distance" begins slowing down when
# nearer than a specified distance. Good if you want to tune collision thresholds manually.
# "stop_distance" stops if a collision is nearer than the worst-case stopping distance and the
# distance is decreasing. Requires joint acceleration limits
collision_check_type: threshold_distance
# Parameters for "threshold_distance"-type collision checking
self_collision_proximity_threshold: 0.01 # Start decelerating when a self-collision is this far [m]
scene_collision_proximity_threshold: 0.02 # Start decelerating when a scene collision is this far [m]
# Parameters for "stop_distance"-type collision checking
collision_distance_safety_factor: 1000.0 # Must be >= 1. A large safety factor is recommended to account for latency
min_allowable_collision_distance: 0.01 # Stop if a collision is closer than this [m]

```

### Expected behavior

End effector should move only on the z-axis

### Actual behavior

The robot doesn't make the correct movement

sea-bass, 2025-01-10:
I think I may have made a mistake in the twist command scaling in #3007 -- could you try this branch and see how it goes? https://github.com/moveit/moveit2/pull/3225

Also generally, your servo configuration file seems to have a lot of deprecated parameters in it. See here for a list of what you actually need:

https://github.com/moveit/moveit2/blob/cd26fc038f0449849cba5084b23d33dfe52ac5a1/moveit_ros/moveit_servo/config/panda_simulated_config.yaml#L1

sea-bass, 2025-01-10:
Those upper/lower singularity threshold values also look pretty high, but I doubt that's the main issue here...

sea-bass, 2025-01-10:
This is a reference UR5 servo configuration which may also help?
https://github.com/PickNikRobotics/moveit_studio_ur_ws/blob/main/src/moveit_pro_ur_configs/picknik_ur_base_config/config/moveit/ur_servo.yaml

PietroValerio, 2025-01-10:
I'll close the issue, modifying the .yaml file resolved the issue. Also this was a problem of ur controllers so not completely related to moveit, thanks for the help!

closed 2025-01-10 by PietroValerio