moveit/moveit2 #3243: Completely lacking documentation on moveit servo

PietroValerio, 2025-01-13:
### Description

Completely lacking documentation, cpp api assumes that the controller used accepts messages of type` trajectory_msgs::msg::joint_trajectory`, what about if we have a controller only accepting `std_msgs/msg/Float64MultiArray`. Nothing inside the documentation covering this use case, you need to go digging inside the codebase to find that there is a function `composeMultiArrayMessage` that maybe can do the job, but you can't know it because is not documented.
You want to be the industry de-facto standard, before you should worry about fixing your documentation

### ROS Distro

Jazzy

### OS and version

24.04

### Source or binary build?

Source

### If source, which branch?

main

### Which RMW are you using?

None

### Steps to Reproduce

Watch moveit servo tutorial

### Expected behavior

Fully documented

### Actual behavior

Lacking documentation

mikeferguson, 2025-01-13:
This is an open source project, staffed by volunteers. PRs to improve documentation are welcome.

attu0, 2025-01-31:
could you check my pr and tell if any changes required

mikeferguson, 2025-01-31:
> could you check my pr and tell if any changes required

I'm not the right person to review moveit servo stuff - it's one of the modules I really haven't worked with at all

sea-bass, 2025-02-01:
> [@mikeferguson](https://github.com/mikeferguson) could you check my pr and tell if any changes required

There are already 2 other reviewers on that PR, with pending comments from a few days back. You should be able to proceed by looking at those existing comments.