moveit/moveit2 #3261: Linking issue with butterworth_filter.h

jodle001, 2025-01-22:
### Description

When I try to include the file 

`#include <moveit/online_signal_smoothing/butterworth_filter.h>`

I am no longer able to compile my project. 

### ROS Distro

Humble

### OS and version

Ubuntu 22.04

### Source or binary build?

Binary

### If binary, which release version?

2.5.7-Alpha

### Which RMW are you using?

FastRTPS

### Steps to Reproduce

I have created a project that is a minimal project to reproduce this issue here:

https://github.com/jodle001/moveit_butterworth_minimal_project.git

In the README.md I have detailed the steps to reproduce this error.

## How To Reproduce

1. clone this repository into some directory 
2. cd to cloned repository directory
3. build image from [test.Dockerfile](test.Dockerfile)
    * `docker build --no-cache -f test.Dockerfile -t ros2:test . `
4. Open prject root in VSCode
5. Open command pallet (ctrl+shift+p)
6. Select `Reopen in Container`
7. Using the terminal now running the docker container inside VSCode, build the project with `colcon build`

### Expected behavior

Files should be linking properly and compiling successfully. 

### Actual behavior

Compiling fails due to line 10 in the butterworth_filter.h file.

### Backtrace or Console output

```bash
Starting >>> butterworth_test_package
--- stderr: butterworth_test_package
In file included from /workspaces/moveit_butterworth_minimal_project/src/butterworth_test_package/include/butterworth_test_package/test_file.hpp:3,
                 from /workspaces/moveit_butterworth_minimal_project/src/butterworth_test_package/src/test_file.cpp:1:
/opt/ros/humble/include/moveit/online_signal_smoothing/butterworth_filter.h:44:10: fatal error: moveit_butterworth_parameters.hpp: No such file or directory
   44 | #include <moveit_butterworth_parameters.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/butterworth_test_package.dir/build.make:76: CMakeFiles/butterworth_test_package.dir/src/test_file.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/butterworth_test_package.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< butterworth_test_package [0.06s, exited with code 2]
```

sea-bass, 2025-01-22:
`generate_parameter_library` just updated recently, could you try the following?

```
#include <moveit_core/moveit_butterworth_parameters.hpp>
```

jodle001, 2025-01-23:
I wanted to add a pull request because changing that worked.  but the comment above the includes here is auto generated. 

Do you know where these are auto generated so I can make the change for a pull request?

```c++
// Auto-generated
#include <moveit_butterworth_parameters.hpp>
#include <moveit/robot_model/robot_model.h>
#include <moveit/online_signal_smoothing/smoothing_base_class.h>
```

Locally I can just make the change manually to get it to work.

sea-bass, 2025-01-24:
> Started pull request here [#3264](https://github.com/moveit/moveit2/pull/3264)

Thank you! I've merged it.

closed 2025-01-24 by sea-bass