moveit/moveit2 #3268: Move_group exits with segmentation fault when receiving a SIGINT

JorgePH, 2025-01-24:
### Description

I am receiving a segmentation fault when trying to close move_group node (with Ctrl+C or sending SIGINT).

### ROS Distro

Jazzy

### OS and version

Ubuntu 24.04.1 LTS

### Source or binary build?

Binary

### If binary, which release version?

2.12.1-1noble.20241228.033527

### Which RMW are you using?

CycloneDDS

### Steps to Reproduce

Launch ur sim moveit demo (or any other package containing move_group):
```shell
ros2 launch ur_simulation_gz ur_sim_moveit.launch.py 
```
Close it with Ctrl+C

### Expected behavior

All process should close with exit code -2.

### Actual behavior

Move_group node returns a exit code -11, and prints a Segmentation fault error:
The complete error is:
```shell
[move_group-8] Stack trace (most recent call last):
[move_group-8] #6    Object "", at 0xffffffffffffffff, in 
[move_group-8] #5    Object "/opt/ros/jazzy/lib/moveit_ros_move_group/move_group", at 0x555b359e8a44, in _start
[move_group-8] #4    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fba90c7d28a, in __libc_start_main
[move_group-8] #3    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fba90c7d1c9, in 
[move_group-8] #2    Object "/opt/ros/jazzy/lib/moveit_ros_move_group/move_group", at 0x555b359e7517, in main
[move_group-8] #1    Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x7fba912546b9, in rclcpp::Executor::~Executor()
[move_group-8] #0    Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x7fba9124d9ce, in 
[move_group-8] Segmentation fault (Address not mapped to object [0x7fba50f35170])
``` 

### Backtrace or Console output

https://gist.github.com/JorgePH/03bff938f42c78cd6256a0b379619db1

JorgePH, 2025-01-24:
This is problematic specially for all test checking exit codes, because move_group makes them fail.

CihatAltiparmak, 2025-03-21:
Hello, I can deal with this issue. AFAIK, I sorted out this by resetting some interface/node instances [here](https://github.com/moveit/moveit_middleware_benchmark/blob/60d8748d4c8848c80b5c22bf30cfb8048e0e7a6c/src/scenario_perception_pipeline/scenario_perception_pipeline.cpp#L92-L101) (please don't take into account the comments which is badly written). It seems to me that the issue can be related to the order of destroying instances. will try to take a look this weekend.

CihatAltiparmak, 2025-04-13:
Hello, @JorgePH  , firstly thank you for feedbacking us about it and sorry for delayed response. I think I have a patch for it. Will clean up it and make it ready to merge tomorrow. When everything is ready, I will want you to give a feedback to me.  Not now. Just for updating this thread.

PR #3435 (2025-04-15): Make the destructors of the base classes of planning adapters virtual and close move_group gracefully
### Description

Fixes #3268 

### Checklist
- [ ] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [ ] Extend the tutorials / documentation [reference](http://moveit.ros.org/documentation/contributing/)
- [ ] Document API changes relevant to the user in the [MIGRATION.md](https://github.com/moveit/moveit2/blob/main/MIGRATION.md) notes
- [ ] Create tests, which fail without this PR [reference](https://moveit.picknik.ai/hum [... cut]
PR #3443 (2025-04-15): Make the destructors of the base classes of planning adapters virtual and close move_group gracefully (backport #3435)
### Description

Fixes #3268 

### Checklist
- [ ] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [ ] Extend the tutorials / documentation [reference](http://moveit.ros.org/documentation/contributing/)
- [ ] Document API changes relevant to the user in the [MIGRATION.md](https://github.com/moveit/moveit2/blob/main/MIGRATION.md) notes
- [ ] Create tests, which fail without this PR [reference](https://move [... cut]
closed 2025-04-15 by sea-bass