moveit/moveit2 #3285: Inflated self robot link removal from Octree

Ayush1285, 2025-01-30:
### Is your feature request related to a problem? Please describe.

I'm using PoinCloudOctomapUpdater plugin for building Octomap of the environment for collision checking. Problem that I'm facing is that there are few tool wires around robotic arm which are getting registered in Octree. I was able to fix it using padding and subsample parameters, but that is affecting quality of whole environment Octomap.

### Describe the solution you'd like

We can declare a parameter called something like robot_link_inflation with default value to be 1.0. Then in PlanningSceneMonitor::excludeRobotLinksFromOctree(), we can inflate the meshes according to this parameter.

Ayush1285, 2025-02-19:
@sea-bass Any opinion on this?

sea-bass, 2025-02-19:
So is the idea that you don't want to inflate your URDF meshes because that would limit your motion range, but you still want the octomap to filter things out a little further from the collision meshes?

I honestly have no experience with this side of MoveIt, so if this is not already possible, it would be a good addition!

The only thing I can suggest in you trying this is to make sure that you're not inflating the meshes repeatedly as this will cause slowdowns. It could either be storing the inflated meshes in memory, or if this tool happens to use distance vs. occupancy computations, simply subtract out the padding. Again, I'm not really sure how this all works.

Maybe someone like @mikeferguson has more useful input here.

Ayush1285, 2025-02-19:
> So is the idea that you don't want to inflate your URDF meshes because that would limit your motion range, but you still want the octomap to filter things out a little further from the collision meshes?

Yes , we will not modify the original mesh itself. But to just filter out the robot links from the octree, we can use inflated mesh or some nearest neighbour based filtering.

mikeferguson, 2025-02-19:
Also, I had to look for a moment - [here are the descriptions of those parameters](https://moveit.picknik.ai/main/doc/examples/perception_pipeline/perception_pipeline_tutorial.html): 

 * padding_offset: The size of the padding (in cm).
 * padding_scale: The scale of the padding.

Ayush1285, 2025-02-19:
> I believe the parameters you are looking for are ["padding_scale" and padding_offset"](https://github.com/moveit/moveit2/blob/main/moveit_ros/perception/pointcloud_octomap_updater/src/pointcloud_octomap_updater.cpp#L81-L82) in the point cloud updater namespace. I use the depth image version on robot - but the parameters are pretty similar and you can [see how my sensors_3d is configured here](https://github.com/mikeferguson/ubr_reloaded/blob/ros2/ubr1_moveit/config/sensors_3d.yaml).

Ah yes, I checked the source code. padding_offset and padding_scale params are passed to the geometric_shapes::Shapes module for padding the mesh. Thanks for the help @mikeferguson . 
We can add a more verbose description for these parameters, as the current descriptions 'padding_offset: The size of the padding (in cm), padding_scale: The scale of the padding' don't explain much.

mikeferguson, 2025-02-19:
> We can add a more verbose description for these parameters, as the current descriptions 'padding_offset: The size of the padding (in cm), padding_scale: The scale of the padding' don't explain much.

Yes, please open a PR if you get a chance

Ayush1285, 2025-04-07:
Closing the issue.

closed 2025-04-07 by Ayush1285