moveit/moveit2 #3286: PointcloudOctomapUpdater not cleaning at max_range

MarcoMagriDev, 2025-01-31:
### Description

PointcloudOctomapUpdater not cleaning voxels close to max_range.

---

After investigating the issue, I believe I have identified its cause and a possible solution.
The issue may arise from the fact that the **same** `max_range` value is used both [here](https://github.com/moveit/moveit2/blob/0af5e0f45447e24f118b6210d5d5e7eb92868245/moveit_ros/perception/pointcloud_octomap_updater/src/pointcloud_octomap_updater.cpp#L240) and [here](https://github.com/moveit/moveit2/blob/0af5e0f45447e24f118b6210d5d5e7eb92868245/moveit_ros/perception/pointcloud_octomap_updater/src/pointcloud_octomap_updater.cpp#L300).  By decoupling these two values, specifically reducing maskContainment as follows:

```shape_mask_->maskContainment(*cloud_msg, sensor_origin_eigen, 0.0, max_range_ - 0.1, mask_);```

The issue appears to be resolved. I suspect that the 0.1 value is related to the octomap_resolution.

If this quick but effective fix seems acceptable, I’d be happy to submit a PR for it. Let me know your thoughts!

### ROS Distro

Humble

### OS and version

Ubuntu 22.04

### Source or binary build?

Source

### If source, which branch?

humble

### Which RMW are you using?

CycloneDDS

### Steps to Reproduce

1. Position the robot in a fixed location so that the camera faces forward.
2. Take an object and move it within the perception range defined by the `max_range` parameter of the plugin.
3. Observe that the generated Octomap points are successfully cleaned up as long as the object remains within `max_range`.
4. Move the object along the boundary of `max_range`, following a circular path centered on the camera frame.
5. You will notice that voxels near max_range are not being cleaned properly.

### Expected behavior

Points at the `max_range` boundaries should be cleaned up 

### Actual behavior

Points at the `max_range` boundary are not being cleaned up.
However, if you slightly move the robot toward the uncleared voxels, they are correctly removed.

sea-bass, 2025-01-31:
The fix would need to handle any octomap resolution, so instead of hard-coding 0.1 it should of course get that information somehow.

Also, a test, or at least some images of what you are seeing before and after the changes, would be useful as well.

MarcoMagriDev, 2025-01-31:
> The fix would need to handle any octomap resolution, so instead of hard-coding 0.1 it should of course get that information somehow.

Of course! I wasn’t suggesting hardcoding that value. After looking into it further, I think `monitor_->getMapResolution()` is the right way to handle this dynamically. What do you think?

> Also, a test, or at least some images of what you are seeing before and after the changes, would be useful as well.

I can try modifying the tutorial scene to highlight this issue better and collect some pictures. Would that be helpful?

sea-bass, 2025-02-01:
I think both your suggestions make sense.

What I am looking for is a way to convince reviewers (who would be more familiar than I am with this pipeline) that modifying the range value passed into `maskContainment()` correctly handles your edge case, assisted with images and/or tests.

MarcoMagriDev, 2025-02-01:
### Setup
I started from the setup proposed [here](https://moveit.picknik.ai/main/doc/examples/perception_pipeline/perception_pipeline_tutorial.html) and made the following modifications to remove unnecessary or potentially confusing components:

- Removed the table from the .world file.
- Removed Camera 2, as it is linked to the DepthImageOctomapUpdater, which is not relevant to this discussion.
- Manually added a cube after starting the Gazebo simulation.

Note that I did not create any .bag file. I'm simply launching the environment and the perception pipeline together.

### Issue Visualization
To demonstrate the issue, I moved the cube’s closest face toward the camera until it reached the `max_range` boundary (5.0 m in this example). Then, I moved the cube tangentially along the circumference defined by `max_range`. As seen in the attached images, the past positions of the collision object remain in the octomap due to the cleaning issue.

[image]
[image]

### Solution Visualization
I then applied the proposed fix to maskContainment and repeated the experiment. The results are shown in the attached images.

[image]
[image]
[image]

---

Does this look good to move forward with a PR for the fix, or would you like me to add anything else?

sea-bass, 2025-02-01:
That's great! It clears things up for me. Thank you.

I would say go for the PR now!

MarcoMagriDev, 2025-02-01:
Totally agree with your last sentence! I believe this approach correctly handles the clipping. Of course, we should keep in mind that the actual range is now the specified `max_range` parameter minus the octomap resolution, but I don’t think this should be a major issue.

I'll go ahead and open the PR now! 🚀

closed 2025-02-04 by sea-bass