moveit/moveit2 #3293: Remote controlling of a UR5e arm using servo limitations

BrocoliAsesino, 2025-02-01:
### Description

Hello community!

I ve been working for the past few weeks on controlling a UR5e remotely and I have reached the point where I don´t know if I am doing something wrong or simply what I am trying to achieve just doesn´t work/doesn´t make sense. 

The idea is that I have an Optitrack that can record the poses of a mock gripper, it is these movements that I want the UR5e to follow. Since the poses are in the Optitrack's absolute frame, before passing them to my implementaion of PoseTracking, I find the difference between consecutive poses, transform it to be in the gripper s frame and publish them to `/target_pose` (poses wrt Tool0).   

The movements done with the mock gripper under the Optitrack's recording are not particulary fast, they are at an average human arm speed. The issue that I am seing is that the highest `/target_pose`  publishing rate is, the movements that I see out of the robot are in the correct direction, but are tremendously scaled down. We are talking publishing rates between 30 to 50 Hz. To compensate that, I need to set very large proportional gains (in the order of 100) however this does not seem appropiate. I am using the forward_position_controller, in Moveit(1) I had the same issue and tried the JointTrajectoryController and forward_velocity_controller with the same results.  

So my question is, is this a problem of the controllers that are limiting the commands published by moveit servo or, is it that when the poses keep changing very often and thus the twist commands are updated frequently, servo fails to generate the appropiate commands. I know servo in itself is quite simple and that PoseTracking is simply a PID, but the fact that increasing the proportional gain makes it track the movements better makes me question if it comes from there.  

Any help would be greatly appreciated!

### ROS Distro

Humble

### OS and version

Ubuntu 22.04

### Source or binary build?

Source

### If source, which branch?

Humble

### Which RMW are you using?

None

### Steps to Reproduce

Only way to reproduce it would be with a UR5e. 

### Expected behavior

Correct tracking of the mock gripper.

### Actual behavior

Highly scaled down behavior.

sea-bass, 2025-02-01:
Do you know if this scaling down is due to hitting joint velocity/acceleration limits, or is the robot within limits but the output is wrong? I believe MoveIt Servo publishes a status enum to help investigate this.

If it's the latter, you may want to try the new version of MoveIt Servo, not the Humble one, as this can directly take pose commands instead of going through the PID controller demo node that converts from poses to twists.

BrocoliAsesino, 2025-02-02:
Hey @sea-bass, the joint limits are well loaded in the `movegroup` node and the movements are far from reaching these limits. I have been able to pinpoint that the error does not come from the controllers, it is from the output of moveit servo itself. The joint positions published in `forward_position_controller/commands` topic have barely any variation when the proportional gain is low . Only when the proportional gain is large, ~100, does the robot move in the expected way.

This brings me to the hypothesis that `moveit servo` is able to track correctly twist commands that are consistent for some time in the same direction, like in the `PoseTracking` example or when visual servoing is performed, which usually involves moving in the direction that minimizes the error but this one usually decreases in every iteration. 

In my case, where the twist commands are published at 30 Hz, the direction and magnituded keeps varying causing servo to track correctly the direction, both in translation and rotation, but tremendously scaled down. This can be solved by forcing the reactivity by either setting very large proportional gains, reducing the publishing frequency to 10 Hz or doing very slow movements with the mock gripper so that servo can keep up. Does this make sense?

sea-bass, 2025-02-02:
Got it -- so it could be based on the robot's provided velocity acceleration limits, or also due to the low-pass filter based "smoothing plugin" that tries to keep some continuity between commands.

I'd normally not advise completely disabling the smoothing plugins when working with real robots, but in the case of a UR5e, their controllers already have their own internal smoothing so you might be able to run that quick test without doing anything catastrophic to your robot :)

BrocoliAsesino, 2025-02-06:
Hey @sea-bass! The filter lead was perfect. 

[image]

What I did was simply comment the line **777** in `moveit_servo/src/servo_calcs.cpp`  that applies the Butterworth filter:
`smoother_->doSmoothing(joint_state.position);`

It is not ideal, however filtering in positions even for a really low value of the filter_coeff (1.5) had a huge impact, more than I expected. I am guessing that this will imply more jerkiness  using the `forward_position_controller`. I wonder if it would be better to finely tune the PID (to convert PoseStamped to TwistStamped) which now I have it at reasonable value  P=10 or use the JointTrajectory controller, knowing that this will filter the joint commands and increase the proportional gain. 

In any case this was a great lead, thanks a lot!

sea-bass, 2025-02-06:
Great to hear!

Between those 2 options, I would say first go for what works best on your end. I might personally bias to *not* having to tune PID gains for an outer loop to compensate for unpredictable outputs in the inner loop.

sea-bass, 2025-02-07:
Oh also, rather than commenting out that line and recompiling, you can just set `use_smoothing: false` in your servo config YAML file.

BrocoliAsesino, 2025-02-10:
@sea-bass didn t know there was that param! Thanks a lot for everything, I will be closing the issue with that answer :)

closed 2025-02-10 by BrocoliAsesino