moveit/moveit2 #3302: `moveit_servo` regression on Rolling Buildfarm

Yadunund, 2025-02-03:
### Description

Binary job on Rolling is failing for `moveit_servo`. See : https://build.ros2.org/view/Rbin_uN64/job/Rbin_uN64__moveit_servo__ubuntu_noble_amd64__binary/

```bash
/tmp/binarydeb/ros-rolling-moveit-servo-2.12.0/src/servo_node.cpp:42:10: fatal error: realtime_tools/thread_priority.hpp: No such file or directory
   42 | #include <realtime_tools/thread_priority.hpp>
```

We will be proceeding with the upcoming [Rolling sync](https://discourse.ros.org/t/preparing-for-rolling-sync-2025-02-03/) which means this package and downstream ones will be removed from Rolling. They will be restored after a new release of `movit2` is bloomed for Rolling with the fixes applied.

### ROS Distro

Rolling

### OS and version

Ubuntu 24.04

### Source or binary build?

Source

### Which RMW are you using?

None

### Steps to Reproduce

Build moveit_servo 2.12.0 from source against latest `realtime_tools`

### Expected behavior

Build completes successfully.

### Actual behavior

Build fails.

### Backtrace or Console output

```bash
/tmp/binarydeb/ros-rolling-moveit-servo-2.12.0/src/servo_node.cpp:42:10: fatal error: realtime_tools/thread_priority.hpp: No such file or directory
   42 | #include <realtime_tools/thread_priority.hpp>
```

sea-bass, 2025-02-03:
Thanks! I think this has already been addressed in https://github.com/moveit/moveit2/pull/3139 but we haven't bloomed Rolling since then.

Was waiting on bigger changes before updating the minor version on Rolling, but looks like we should do it sooner.

sea-bass, 2025-02-03:
By the way, who gets alerted about these breakages? I could have taken care of this months ago, but it seems the builds only started failing 2 days ago.

The deprecation warnings must have been removed from the realtime_tools right before the sync.

Yadunund, 2025-02-04:
Sounds good! Will leave this ticket open until the release is bloomed!

sea-bass, 2025-02-05:
BTW for people still at PickNik who may be seeing this, the `picknik_controllers` repo also got kicked off Rolling due to other deprecations in the `realtime_tools` package -- this, in turn, impacts the Kinova Gen3 MoveIt packages that also went away this sync.

Please merge https://github.com/PickNikRobotics/picknik_controllers and cut a release for the next sync as well -- thanks!

urfeex, 2025-02-05:
@sea-bass Updating the maintainers in the `package.xml` to those taking care about the packages would probably be a good idea. As far as I know, those are the ones being notified in any case.

Appreciating the quick action to get our `ur_moveit_config`, `ur` and `ur_simulation_gz` packages back in :-)

urfeex, 2025-02-05:
> I see the dependency on `moveit_servo` in the `ur_moveit_config` package -- which transitively affects the other 2 you mentioned. Do you also depend on `picknik_controllers` on the driver side?

No, we don't depend on that. If `moveit_servo` was available, that would fix our issues at the moment :-)

christophfroehlich, 2025-02-06:
Sorry guys, this was my fault. I thought it was sufficient that the deprecation warnings were released for more than 2 months, but they were only included in he last sync in January :/
And it was unfortunate to bloom the breaking release shortly before the sync, which immediately got merged.

Thanks for clarification of the notifications, I thought also that the maintainer of the breaking release get notified if several downstream packages are broken. But this might not be the case?

sea-bass, 2025-02-06:
All is fine @christophfroehlich! I think a big point of the buildfarm is to force people to update, and anything goes on Rolling.

 I'm just explaining the reasons so others can understand why, that the fix is easy and should be applied soon, and how to fix their own packages if they haven't done so.

urfeex, 2025-02-13:
@sea-bass any estimate on when a new version will be released?

sea-bass, 2025-02-13:
There's 1 more bugfix PR I am waiting for (https://github.com/moveit/moveit2/pull/3341), so hoping for the release to be lined up in rosdistro this weekend.

sea-bass, 2025-02-16:
Alright -- MoveIt is queued up for Rolling now: https://github.com/ros/rosdistro/pull/44487

Thanks for your patience. I didn't think the next Rolling sync would just be 2 weeks apart, so we also missed the February 14 one... but should be good for next one!

sea-bass, 2025-03-07:
Servo and the UR packages are back on Rolling! Closing this.

closed 2025-03-07 by sea-bass