moveit/moveit2 #3312: The complete state of the robot is not yet known

tajish, 2025-02-05:
Hello,
I was working with the motion planning of ur5 robot through using rviz. I moveit controller was imported by moveit2. But whenever I try to launch those files, it gives me error like:-

The complete state of the robot is not yet known. Missing robotiq_85_left_finger_joint, robotiq_85_right_finger_joint.

I am attaching the srdf file for your convinience. Can someone help me with that.

<?xml version="1.0" encoding="UTF-8"?>
<robot name="ur5_robot">
    <group name="manipulator">
        <link name="shoulder_link"/>
        <link name="upper_arm_link"/>
        <link name="forearm_link"/>
        <link name="wrist_1_link"/>
        <link name="wrist_2_link"/>
        <link name="wrist_3_link"/>
        <joint name="shoulder_pan_joint"/>
        <joint name="shoulder_lift_joint"/>
        <joint name="elbow_joint"/>
        <joint name="wrist_1_joint"/>
        <joint name="wrist_2_joint"/>
        <joint name="wrist_3_joint"/>
    </group>
    <group name="gripper">
        <link name="robotiq_85_base_link"/>
        <link name="camera_link"/>
        <link name="camera_frame_link"/>
        <link name="robotiq_85_left_inner_knuckle_link"/>
        <link name="robotiq_85_left_finger_tip_link"/>
        <link name="robotiq_85_left_knuckle_link"/>
        <link name="robotiq_85_left_finger_link"/>
        <link name="robotiq_85_right_inner_knuckle_link"/>
        <link name="robotiq_85_right_finger_tip_link"/>
        <link name="robotiq_85_right_knuckle_link"/>
        <link name="robotiq_85_right_finger_link"/>
        <joint name="robotiq_85_left_knuckle_joint"/>
    </group>
    
    <group_state name="zero" group="manipulator">
        <joint name="elbow_joint" value="0"/>
        <joint name="shoulder_lift_joint" value="0"/>
        <joint name="shoulder_pan_joint" value="0"/>
        <joint name="wrist_1_joint" value="0"/>
        <joint name="wrist_2_joint" value="0"/>
        <joint name="wrist_3_joint" value="0"/>
    </group_state>
    <group_state name="up" group="manipulator">
        <joint name="elbow_joint" value="0"/>
        <joint name="shoulder_lift_joint" value="-1.57"/>
        <joint name="shoulder_pan_joint" value="-1.57"/>
        <joint name="wrist_1_joint" value="0"/>
        <joint name="wrist_2_joint" value="0"/>
        <joint name="wrist_3_joint" value="0"/>
    </group_state>
    <group_state name="open" group="gripper">
        <joint name="robotiq_85_left_finger_joint" value="0"/>
        <joint name="robotiq_85_left_knuckle_joint" value="0"/>
        <joint name="robotiq_85_right_finger_joint" value="0"/>
    </group_state>
    <group_state name="close" group="gripper">
        <joint name="robotiq_85_left_finger_joint" value="0"/>
        <joint name="robotiq_85_left_knuckle_joint" value="0.804"/>
        <joint name="robotiq_85_right_finger_joint" value="0"/>
    </group_state>
    
    <end_effector name="robotiq_gripper" parent_link="tool0" group="gripper"/>
    
    <virtual_joint name="virtual_joint" type="fixed" parent_frame="world" child_link="base_link"/>
    
    <passive_joint name="robotiq_85_left_finger_joint"/>
[... 55 lines cut ...]
    <disable_collisions link1="robotiq_85_left_knuckle_link" link2="wrist_1_link" reason="Never"/>
    <disable_collisions link1="robotiq_85_left_knuckle_link" link2="wrist_2_link" reason="Never"/>
    <disable_collisions link1="robotiq_85_left_knuckle_link" link2="wrist_3_link" reason="Never"/>
    <disable_collisions link1="robotiq_85_right_finger_link" link2="robotiq_85_right_finger_tip_link" reason="Default"/>
    <disable_collisions link1="robotiq_85_right_finger_link" link2="robotiq_85_right_knuckle_link" reason="Adjacent"/>
    <disable_collisions link1="robotiq_85_right_finger_link" link2="wrist_1_link" reason="Never"/>
    <disable_collisions link1="robotiq_85_right_finger_link" link2="wrist_2_link" reason="Never"/>
    <disable_collisions link1="robotiq_85_right_finger_link" link2="wrist_3_link" reason="Never"/>
    <disable_collisions link1="robotiq_85_right_finger_tip_link" link2="robotiq_85_right_inner_knuckle_link" reason="Adjacent"/>
    <disable_collisions link1="robotiq_85_right_finger_tip_link" link2="robotiq_85_right_knuckle_link" reason="Never"/>
    <disable_collisions link1="robotiq_85_right_finger_tip_link" link2="wrist_1_link" reason="Never"/>
    <disable_collisions link1="robotiq_85_right_finger_tip_link" link2="wrist_2_link" reason="Never"/>
    <disable_collisions link1="robotiq_85_right_finger_tip_link" link2="wrist_3_link" reason="Never"/>
    <disable_collisions link1="robotiq_85_right_inner_knuckle_link" link2="robotiq_85_right_knuckle_link" reason="Never"/>
    <disable_collisions link1="robotiq_85_right_inner_knuckle_link" link2="wrist_1_link" reason="Never"/>
    <disable_collisions link1="robotiq_85_right_inner_knuckle_link" link2="wrist_2_link" reason="Never"/>
    <disable_collisions link1="robotiq_85_right_inner_knuckle_link" link2="wrist_3_link" reason="Never"/>
    <disable_collisions link1="robotiq_85_right_knuckle_link" link2="wrist_1_link" reason="Never"/>
    <disable_collisions link1="robotiq_85_right_knuckle_link" link2="wrist_2_link" reason="Never"/>
    <disable_collisions link1="robotiq_85_right_knuckle_link" link2="wrist_3_link" reason="Never"/>
    <disable_collisions link1="shoulder_link" link2="upper_arm_link" reason="Adjacent"/>
    <disable_collisions link1="wrist_1_link" link2="wrist_2_link" reason="Adjacent"/>
    <disable_collisions link1="wrist_1_link" link2="wrist_3_link" reason="Never"/>
    <disable_collisions link1="wrist_2_link" link2="wrist_3_link" reason="Adjacent"/>
</robot>

sea-bass, 2025-02-05:
Your SRDF file mentioned a bunch of robotiq gripper grounds, but does your URDF file include a robotiq gripper? From the limited information above, it indicates that maybe you are launching with only the UR arm?

tajish, 2025-02-06:
@sea-bass hello I was able to fix that error but now the problem is I can't plan and execute any goal by using moveit2 in rviz. It fails. And says not enough state found for the goal.

https://drive.google.com/drive/folders/1z1JulbjFv69nY9MVFNonjKz0Ih5hEwAV?usp=sharing

If you dont mind can you have a look at my files ,where did I made mistake. It would be very helpful. Thank you

Danilrivero, 2025-02-10:
Rather seems to be something happening to the controller of the gripper, manually verify that you are able to control the UR and gripper separately first. Check [this](https://github.com/PickNikRobotics/robotiq_85_gripper) for more information that might be useful. Specially the last part [here](https://github.com/PickNikRobotics/robotiq_85_gripper#issues) (in case that's your gripper)

tajish, 2025-02-11:
@Danilrivero thank you for the advice.Eventually I was able to solve the problem , it was a problem related to the joint limits of the joints where I didn't specify the velocity and the max acc of some joints, for which the controllers weren't taking commands. 

And talking about the last part you have refered, I don't think robotiq_85_right_knuckle_joint should be in the group state of the gripper as it will mimic robotiq_85_left_knuckle_joint.

gyw0708, 2025-04-06:
> [@sea-bass](https://github.com/sea-bass) hello I was able to fix that error but now the problem is I can't plan and execute any goal by using moveit2 in rviz. It fails. And says not enough state [...]

How did u manage to fix the error of the robotiq joint?