moveit/moveit2 #3339: (Humble) Interactive Markers do not display

rr-tom-noble, 2025-02-12:
### Description

We've been debugging an issue where interactive markers do not display when using the Query Goal State toggle in the Motion Planning Display > Planning Request section. This is also accompanied by a warning message from rviz:

```
[rviz2-1] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
```

Removing the MotionPlanningDisplay from our setup removes the warning, and manually adding the Display back in through the Rviz UI causes the warning to trigger again.

I wasn't sure if this was an issue specific to our setup, so I tried the same thing using the `moveit/moveit2:main-humble-tutorial-source` image, with the same results, detailed in the reproduction steps section.

### ROS Distro

Humble

### OS and version

Ubuntu 22.04

### Source or binary build?

Binary

### If binary, which release version?

moveit2:main-humble-tutorial-source Docker image (without building any in-container code)

### Which RMW are you using?

None

### Steps to Reproduce

Follow steps to download and run the tutorial image, then turn on the Query Goal State option under the Motion Planning Display:

```
wget https://raw.githubusercontent.com/moveit/moveit2_tutorials/main/.docker/docker-compose.yml
DOCKER_IMAGE=main-humble-tutorial-release docker compose run --rm --name moveit2_container cpu
(Container) source /opt/ros/humble/setup.bash && ros2 launch moveit2_tutorials demo.launch.py
```

### Expected behavior

When Query Goal State is enabled, an interactive marker should appear at the tip of the robot, allowing it to be moved around to set the goal state.

The following image from https://moveit.picknik.ai/main/doc/tutorials/quickstart_in_rviz/quickstart_in_rviz_tutorial.html shows the marker:

[image]

### Actual behavior

No such interactive marker appears:

[image]

rr-tom-noble, 2025-02-12:
Reproduced with the jazzy version of the tutorial image too. Will try with rolling next.

rr-tom-noble, 2025-02-12:
Ah. Setting the Planning Group to "manipulator" rather than "gripper" in the demo causes the markers to appear. I'm not entirely sure what the conditions are for determining whether a planning group should display the markers, but it seems like this is the issue (may be that we need to tweak our setup a bit)

gavanderhoorn, 2025-02-12:
Only groups with a configured IK solver will show the marker, as otherwise no IK is possible.

> may be that we need to tweak our setup a bit

if you define a chain with an IK solver and additionally an EEF where that group is the parent, things should-just-work™.

sea-bass, 2025-02-12:
This is a weird "feature" that has annoyed me for a while too. Can you change the "Interactive Marker Size" field to a nonzero value like 0.25?

rr-tom-noble, 2025-02-12:
> This is a weird "feature" that has annoyed me for a while too. Can you change the "Interactive Marker Size" field to a nonzero value like 0.25?

Just tried this with no luck 😞 

It's worth noting that most of our robot config (xacros, launchfiles, etc...) were pretty hastily ported over to ROS2. Although we did make a few changes to get moveit planning working, maybe there's still some outdated syntax causing some information to not propagate to the rviz plugin...

I'll have a poke around in the moveit2 demo configs and see if I can spot any significant differences.

rr-tom-noble, 2025-02-12:
Got some more helpful logs after tinkering with the logging level:

```
[rviz2-3] [INFO] [1739373845.650344023] [moveit_ros_robot_interaction.robot_interaction]: No active joints or end effectors found for group 'gp25_only'. Make sure that kinematics.yaml is loaded in this node's namespace.
[rviz2-3] [INFO] [1739373845.650962449] [moveit_ros_visualization.motion_planning_frame]: group gp25_only
[rviz2-3] [INFO] [1739373845.650977150] [moveit_ros_visualization.motion_planning_frame]: Constructing new MoveGroup connection for group 'gp25_only' in namespace ''
[rviz2-3] [INFO] [1739373845.671784997] [move_group_interface]: Ready to take commands for planning group gp25_only.
```

We do pass the kinematics into the move group node via the "robot_description_kinematics" parameter. We've had issues where rviz doesn't pick up on topics published by moveit despite being in the same namespace, forcing us to add a bunch of remappings. I'm wondering if there's something similar happening with the kinematics info.

rr-tom-noble, 2025-02-13:
I've managed to fix the issue by explicitly passing the "robot_description_kinematics" parameter to the rviz node in addition to the move group node. 

I've also found that setting the "Move Group Namespace" in the rviz settings file allows us to remove all of the re-mappings we do in the rviz node, though I'm still a little unsure why this is necessary, since we launch the move group and rviz nodes in the same namespace ("/move_group")

I thought that adding the "Move Group Namespace" might also fix the kinematics info not being available to rviz, but this is not the case, though using both of these solutions together causes a crash with the following error:

```
[rviz2-3] [ERROR] [1739437454.350926710] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': logger with name 'rviz_gp25_only_ik_plugin' already exists
```

@gavanderhoorn Do you know if the move group should be passing kinematics information to the rviz plugin (e.g. via a topic or service), provided the namespacing is correct? It'd be nice if we could remove all of our gross remappings (which are likely not exhaustive) and fix the display / rviz kinematics issue at the same time.

gavanderhoorn, 2025-02-13:
> I've managed to fix the issue by explicitly passing the "robot_description_kinematics" parameter to the rviz node in addition to the move group node.

ah, so you had no IK plugins defined as far as the RViz plugin was concerned. That would indeed cause the IM not to show up.

> Do you know if the move group should be passing kinematics information to the rviz plugin (e.g. via a topic or service), provided the namespacing is correct? 

unless something has changed recently: no, the RViz plugin needs its own set of parameters. It doesn't retrieve them from the move group node.

sea-bass, 2025-02-13:
Thanks @gavanderhoorn and @Danilrivero!

I guess a lot of this is likely a byproduct of ROS 2 not having global-scoped parameters by design, so now there are big overlapping sets of duplicated parameters that need to be appropriately delegated to each node at the launch file level.

Grabbing this information from topics instead -- or services, which is how parameters work under the hood -- would be a pattern closer to ROS 1. It also can be nicer for synchronization since the "source of truth" is stored in one node.

Ultimately, my opinion tracks the "zen of Python" -- there should be one obvious, unambiguous way of doing things, if at all possible.

sea-bass, 2025-02-13:
I also think that, with the current design, a few log messages to help debug would go a long way.

For example, if the MoveIt RViz widget spit out a warning like "No kinematics information found, did you include XYZ?", it could already go a long way IMO.

rr-tom-noble, 2025-02-13:
Thanks all! I'm still a little unsure why using a Move Group Namespace with the kinematics parameter causes a logger name collision & crash, however, this is less related to the original issue which is now fixed, so I'll close this ticket 👍

closed 2025-02-13 by rr-tom-noble