moveit/moveit2 #3340: [move_group_interface]: Failed to fetch current robot state

LucaMIMOUNI, 2025-02-12:
### Description

Hey,

## System
* Ubuntu 22.04 Jammy
* Ros2 Humble

## Error

```
[INFO] [1739369238.330720489] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[INFO] [1739369239.330854109] [moveit_ros.current_state_monitor]: Didn't receive robot state (joint angles) with recent timestamp within 1.000000 seconds. Requested time 1739369238.330760, but latest received state has time 0.000000.
Check clock synchronization if your are running ROS across multiple machines!
[ERROR] [1739369239.330916169] [move_group_interface]: Failed to fetch current robot state
```

## Context
Initially, I was just trying to get the State of my ur10e (from the Gazebo simu) with moveit by following this tutorial [moveit_tuto](https://moveit.picknik.ai/main/doc/examples/move_group_interface/move_group_interface_tutorial.html)
However, I face the error above with ` *move_group_interface.getCurrentState()
`
I've saw a lot of bug report about this subject, suggesting for the most to put `use_sim_time` as `true` for every node, however this doesn't solve the error for me.
I also see this [closed_topic](https://github.com/moveit/moveit2/issues/496#issuecomment-1433122585) about the exact same problem, I've take the code but ran through several errors, especially when simply trying to colcon build the package:

```
CMake Error at /opt/ros/humble/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:77 (message):
  ament_target_dependencies() the passed package name
  'hello_moveit__rosidl_typesupport_cpp' was not found before
Call Stack (most recent call first):
  CMakeLists.txt:31 (ament_target_dependencies)
```

Any Idea of what is happening here, kind of confuse with the services.

Here is my `CMakeList.txt`:

```
cmake_minimum_required(VERSION 3.8)
project(hello_moveit)

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

# Find dependencies
find_package(ament_cmake REQUIRED)
find_package(moveit_ros_planning_interface REQUIRED)
find_package(rclcpp REQUIRED)
find_package(rosidl_default_generators REQUIRED)
find_package(geometry_msgs REQUIRED)

# Generate service files
rosidl_generate_interfaces(${PROJECT_NAME}
  "srv/ExampleService.srv"
  DEPENDENCIES geometry_msgs
)

# Define the executable target
add_executable(move_robot_node src/move_robot.cpp)
ament_target_dependencies(move_robot_node
  "moveit_ros_planning_interface"
  "rclcpp"
  "geometry_msgs"
  "rosidl_default_runtime"  # Ensure the runtime is linked
)

# Link against the generated type support library
ament_target_dependencies(move_robot_node
  "${PROJECT_NAME}__rosidl_typesupport_cpp"
)

# Include directories and compile features
target_include_directories(move_robot_node PUBLIC
  $<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}/include>
  $<INSTALL_INTERFACE:include>
  $<BUILD_INTERFACE:${CMAKE_CURRENT_BINARY_DIR}/rosidl_generator_cpp>  # Include generated files
)
target_compile_features(move_robot_node PUBLIC c_std_99 cxx_std_17)

# Install the executable
install(TARGETS move_robot_node
  DESTINATION lib/${PROJECT_NAME})

# Install service files
install(DIRECTORY srv/
  DESTINATION share/${PROJECT_NAME})

if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)
  set(ament_cmake_copyright_FOUND TRUE)
  set(ament_cmake_cpplint_FOUND TRUE)
  ament_lint_auto_find_test_dependencies()
endif()

ament_package()

```

And here is the `ExampleService.srv` : 
```
# ExampleService.srv
geometry_msgs/Transform transform  # Request: Transformation data
---
bool success                       # Response: Indicates whether the action was successful

```

Best !

Luca

### ROS Distro

Humble

### OS and version

Ubuntu 22.04

### Source or binary build?

Source

### Which RMW are you using?

None

### Steps to Reproduce

`move_robot.h`
```
#ifndef MOVE_ROBOT_H
#define MOVE_ROBOT_H

#include <moveit/move_group_interface/move_group_interface.h>
// INCLUDE YOUR SRVS 
#include <hello_moveit/srv/example_service.hpp>

class MoveRobot : public rclcpp::Node {
public:
    explicit MoveRobot(const rclcpp::NodeOptions &options);

private:
    rclcpp::Service<hello_moveit::srv::ExampleService>::SharedPtr service_example_server_;
    std::string node_namespace_;
    moveit::planning_interface::MoveGroupInterfacePtr move_group_;
    rclcpp::Node::SharedPtr node_;
    rclcpp::Executor::SharedPtr executor_;
    std::thread executor_thread_;

    void service_example_callback(const std::shared_ptr<hello_moveit::srv::ExampleService::Request> request,
                                  std::shared_ptr<hello_moveit::srv::ExampleService::Response> response);
};

#endif //YOUR_PACKAGE_YOUR_HEADER_H
```

`move_robot.cpp`
```
#include "hello_moveit/move_robot.h"

static const double PLANNING_TIME_S = 5.0;
static const double MAX_VELOCITY_SCALE = 1.0;
static const double MAX_ACCELERATION_SCALE = 1.0;
static const unsigned int PLANNING_ATTEMPTS = 5;
static const double GOAL_TOLERANCE = 1e-3;

MoveRobot::MoveRobot(const rclcpp::NodeOptions &options)
        : Node("move_robot_node", options), node_(std::make_shared<rclcpp::Node>("example_group_node")),
          executor_(std::make_shared<rclcpp::executors::SingleThreadedExecutor>()) {
    node_namespace_ = ((std::string) this->get_namespace()).erase(0, 1);
    
    service_example_server_ = this->create_service<hello_moveit::srv::ExampleService>(
            "~/example_service",
            std::bind(&MoveRobot::service_example_callback, this, std::placeholders::_1, std::placeholders::_2)
    );

    auto mgi_options = moveit::planning_interface::MoveGroupInterface::Options(
            node_namespace_ + "_ur_manipulator",
            "/" + node_namespace_,
            "robot_description");
    move_group_ = std::make_shared<moveit::planning_interface::MoveGroupInterface>(node_, mgi_options);
    //move_group_->setPoseReferenceFrame("base_link_inertia");
    move_group_->setPlanningTime(PLANNING_TIME_S);
    move_group_->setNumPlanningAttempts(PLANNING_ATTEMPTS);
    move_group_->setGoalTolerance(GOAL_TOLERANCE);
    move_group_->setMaxVelocityScalingFactor(MAX_VELOCITY_SCALE);
    move_group_->setMaxAccelerationScalingFactor(MAX_ACCELERATION_SCALE);
    executor_->add_node(node_);
    executor_thread_ = std::thread([this]() { this->executor_->spin(); });
}

void MoveRobot::service_example_callback(const std::shared_ptr<hello_moveit::srv::ExampleService::Request> request,
                                         std::shared_ptr<hello_moveit::srv::ExampleService::Response> response) {
    move_group_->setStartStateToCurrentState();
    geometry_msgs::msg::Pose pose_goal = move_group_->getCurrentPose().pose;
    pose_goal.position.x += request->transform.translation.x;
    pose_goal.position.y += request->transform.translation.y;
    pose_goal.position.z += request->transform.translation.z;
    tf2::Quaternion q_orig(pose_goal.orientation.x, pose_goal.orientation.y, pose_goal.orientation.z,
                           pose_goal.orientation.w);
    tf2::Quaternion q_rot(request->transform.rotation.x, request->transform.rotation.y, request->transform.rotation.z,
                          request->transform.rotation.w);
    tf2::Quaternion q_new;
    q_new = q_rot * q_orig;
    q_new.normalize();
    pose_goal.orientation.x = q_new.x();
    pose_goal.orientation.y = q_new.y();
    pose_goal.orientation.z = q_new.z();
    pose_goal.orientation.w = q_new.w();
    move_group_->setPoseTarget(pose_goal);
    moveit::planning_interface::MoveGroupInterface::Plan my_plan;
    RCLCPP_INFO(this->get_logger(), "Planned position x: %f, y: %f, z: %f", pose_goal.position.x, pose_goal.position.y,
                pose_goal.position.z);
    bool success = (move_group_->plan(my_plan) == moveit::core::MoveItErrorCode::SUCCESS);
    response->success = success;
    move_group_->move();
}

int main(int argc, char **argv) {
    rclcpp::init(argc, argv);
    rclcpp::NodeOptions node_options;
    node_options.automatically_declare_parameters_from_overrides(true);
    auto move_group_node = std::make_shared<MoveRobot>(node_options);
    rclcpp::spin(move_group_node);

    rclcpp::shutdown();
    return 0;
}

```

`CMakeList.txt` and `ExampleService.srv` above, and then `colcon build --symllink-install`

### Expected behavior

Colcon build right

### Actual behavior

Doesn't colcon build right

sea-bass, 2025-02-12:
I think this is because your constructor requires ROS message data, but you're not spinning the executor until after the constructor is completed.

LucaMIMOUNI, 2025-02-12:
Hey @sea-bass, Thank you for your answer; I'm not so familiar with executor, could you develop a little more ? 

From what I understand I should launch the executor before spinning the node ?

sea-bass, 2025-02-12:
Actually, looking at the code snippet what may be happening is that there are 2 executors competing.

The constructor has

```cpp
executor_->add_node(node_);
    executor_thread_ = std::thread([this]() { this->executor_->spin(); });
```

But the main has

```
rclcpp::spin(move_group_node)
```

Which puts the node into a different (global) executor.

Try removing one of these, maybe the one in the constructor, and see how it goes?

And generally, I would point you to the tutorials for move group usage: https://github.com/moveit/moveit2_tutorials/blob/main/doc/tutorials/planning_around_objects/hello_moveit.cpp

sea-bass, 2025-02-12:
Also regarding your service building failures, I think the ROS 2 build system struggles to resolve message definitions and other code that uses it in the same project.

It is a general pattern to have your message definitions in their own separate package, and then depend on/link against that package in your other package(s) with the source code that may use those definitions.

LucaMIMOUNI, 2025-02-15:
Hey ! So I find the culprit; it was indeed a problem with the `use_sim_time`. The node needed to be run via a launch file so I could specify the correct parameters. 

`hello_moveit.launch.py`

```python
import os

from launch_ros.actions import Node
from launch_ros.substitutions import FindPackageShare
from ur_moveit_config.launch_common import load_yaml

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, OpaqueFunction
from launch.conditions import IfCondition
from launch.substitutions import (
    Command,
    FindExecutable,
    LaunchConfiguration,
    PathJoinSubstitution,
    OrSubstitution,
)

def generate_launch_description():
    robot_description_kinematics = PathJoinSubstitution(
        [FindPackageShare("sr_moveit_config"), "config", "kinematics.yaml"]
    )
    return LaunchDescription([
        Node(
            package='hello_moveit',
            executable='hello_moveit',
            name='hello_moveit',
            output='screen',
            parameters=[
                robot_description_kinematics,
                {'use_sim_time': True}]
        ),
    ])
```
The important thing to note here is the `{'use_sim_time': True}`.

PS : I also remove one of the executor, it wasn't the culprit but it helped to make the code clearer ty !

closed 2025-02-15 by LucaMIMOUNI