moveit/moveit2 #3352: Error while building a package in Docker

Abhishek8857, 2025-02-17:
### Description

`-----                                                                                               
 > [27/27] RUN source /colcon_ws/install/setup.bash &&     colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release:                                           
0.755 Starting >>> agent_listener                                                                    
0.769 Starting >>> base_node_python                                                                  
0.781 Starting >>> kinova_python                                                                     
0.794 Starting >>> robot_actions
0.805 Starting >>> robot_launch
1.730 Finished <<< base_node_python [0.96s]
2.087 Finished <<< kinova_python [1.31s]
2.349 Finished <<< robot_launch [1.54s]
9.165 Finished <<< robot_actions [8.37s]
18.46 --- stderr: agent_listener
18.46 /usr/bin/ld: warning: libgeometric_shapes.so.2.3.1, needed by /colcon_ws/install/moveit_ros_planning/lib/libmoveit_planning_scene_monitor.so.2.5.7, not found (try using -rpath or -rpath-link)
18.46 /usr/bin/ld: warning: libsrdfdom.so.2.0.5, needed by /colcon_ws/install/moveit_core/lib/libmoveit_robot_model.so.2.5.7, not found (try using -rpath or -rpath-link)
18.46 /usr/bin/ld: /colcon_ws/install/moveit_core/lib/libmoveit_robot_model.so.2.5.7: undefined reference to `shapes::createMeshFromResource(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)'
18.46 /usr/bin/ld: /colcon_ws/install/moveit_core/lib/libmoveit_planning_scene.so.2.5.7: undefined reference to `shapes::constructShapeFromText(std::istream&)'
18.46 /usr/bin/ld: /colcon_ws/install/moveit_core/lib/libmoveit_kinematic_constraints.so.2.5.7: undefined reference to `shapes::constructMarkerFromShape(shapes::Shape const*, visualization_msgs::msg::Marker_<std::allocator<void> >&, bool)'
18.46 /usr/bin/ld: /colcon_ws/install/moveit_core/lib/libmoveit_robot_model.so.2.5.7: undefined reference to `typeinfo for shapes::Mesh'
18.46 /usr/bin/ld: /colcon_ws/install/moveit_core/lib/libmoveit_kinematic_constraints.so.2.5.7: undefined reference to `bodies::createEmptyBodyFromShapeType(shapes::ShapeType const&)'
18.46 /usr/bin/ld: /colcon_ws/install/moveit_core/lib/libmoveit_planning_scene.so.2.5.7: undefined reference to `shapes::saveAsText(shapes::Shape const*, std::ostream&)'
18.46 /usr/bin/ld: /colcon_ws/install/moveit_core/lib/libmoveit_robot_model.so.2.5.7: undefined reference to `shapes::Sphere::Sphere(double)'
18.46 /usr/bin/ld: /colcon_ws/install/moveit_core/lib/libmoveit_kinematic_constraints.so.2.5.7: undefined reference to `shapes::constructShapeFromMsg(shape_msgs::msg::SolidPrimitive_<std::allocator<void> > const&)'
18.46 /usr/bin/ld: /colcon_ws/install/moveit_core/lib/libmoveit_kinematic_constraints.so.2.5.7: undefined reference to `shapes::Mesh::Mesh()'
18.46 /usr/bin/ld: /colcon_ws/install/moveit_core/lib/libmoveit_robot_model.so.2.5.7: undefined reference to `shapes::Box::Box(double, double, double)'
18.46 /usr/bin/ld: /colcon_ws/install/moveit_core/lib/libmoveit_robot_model.so.2.5.7: undefined reference to `typeinfo for shapes::Shape'
18.46 /usr/bin/ld: /colcon_ws/install/moveit_core/lib/libmoveit_robot_state.so.2.5.7: undefined reference to `shapes::computeShapeExtents(shapes::Shape const*)'
18.46 /usr/bin/ld: /colcon_ws/install/moveit_core/lib/libmoveit_robot_state.so.2.5.7: undefined reference to `shapes::Shape::padd(double)'
18.46 /usr/bin/ld: /colcon_ws/install/moveit_core/lib/libmoveit_kinematic_constraints.so.2.5.7: undefined reference to `shapes::constructShapeFromMsg(shape_msgs::msg::Mesh_<std::allocator<void> > const&)'
18.46 /usr/bin/ld: /colcon_ws/install/moveit_core/lib/libmoveit_robot_state.so.2.5.7: undefined reference to `shapes::Shape::scale(double)'
18.46 /usr/bin/ld: /colcon_ws/install/moveit_core/lib/libmoveit_robot_model.so.2.5.7: undefined reference to `srdf::Model::empty_vector_'
18.46 /usr/bin/ld: /colcon_ws/install/moveit_core/lib/libmoveit_robot_model.so.2.5.7: undefined reference to `shapes::Cylinder::Cylinder(double, double)'
18.46 /usr/bin/ld: /colcon_ws/install/moveit_core/lib/libmoveit_planning_scene.so.2.5.7: undefined reference to `shapes::OcTree::OcTree(std::shared_ptr<octomap::OcTree const> const&)'
18.46 /usr/bin/ld: /colcon_ws/install/moveit_ros_planning/lib/libmoveit_rdf_loader.so.2.5.7: undefined reference to `srdf::Model::initString(urdf::ModelInterface const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
18.46 /usr/bin/ld: /colcon_ws/install/moveit_core/lib/libmoveit_robot_state.so.2.5.7: undefined reference to `shapes::constructMsgFromShape(shapes::Shape const*, boost::variant<shape_msgs::msg::SolidPrimitive_<std::allocator<void> >, shape_msgs::msg::Mesh_<std::allocator<void> >, shape_msgs::msg::Plane_<std::allocator<void> > >&)'
18.46 /usr/bin/ld: /colcon_ws/install/moveit_core/lib/libmoveit_collision_detection.so.2.5.7: undefined reference to `typeinfo for shapes::OcTree'
18.46 /usr/bin/ld: /colcon_ws/install/moveit_ros_planning/lib/libmoveit_planning_scene_monitor.so.2.5.7: undefined reference to `shapes::Mesh::mergeVertices(double)'
18.46 /usr/bin/ld: /colcon_ws/install/moveit_core/lib/libmoveit_robot_state.so.2.5.7: undefined reference to `shapes::constructShapeFromMsg(shape_msgs::msg::Plane_<std::allocator<void> > const&)'
18.46 collect2: error: ld returned 1 exit status
18.46 gmake[2]: *** [CMakeFiles/agent_listener.dir/build.make:355: agent_listener] Error 1
18.46 gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/agent_listener.dir/all] Error 2
18.46 gmake: *** [Makefile:146: all] Error 2
18.46 ---
18.46 Failed   <<< agent_listener [17.7s, exited with code 2]
18.49 
18.49 Summary: 4 packages finished [17.9s]
18.49   1 package failed: agent_listener
18.49   1 package had stderr output: agent_listener
------
Dockerfile:3`

### ROS Distro

Humble

### OS and version

Ubuntu 22.04

### Source or binary build?

Source

### If source, which branch?

humble

### Which RMW are you using?

[... 103 lines cut ...]
# Copy cyclonedds config files
COPY cyclonedds/config.xml /config.xml
COPY cyclonedds/10-cyclone-max.conf /etc/sysctl.d/10-cyclone-max.conf

# Copy entrypoint scripts and make them executable
COPY entrypoint_scripts/ /entrypoint_scripts/
RUN chmod +x /entrypoint_scripts/*.sh
RUN apt-get update && apt-get upgrade -y

# Copy contents in overlay ws
COPY overlay_ws/ /overlay_ws/
WORKDIR /overlay_ws/

RUN rosdep install --from-paths src --ignore-src -r -y

RUN source /colcon_ws/install/setup.bash && \
    colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release

### Expected behavior

the packages in overlay_ws should build without errors

### Actual behavior

Package failed with errors

sea-bass, 2025-02-17:
This specifically is pointing to the `geometric_shapes` repo having a linking issue.

Are you building it from source or installing it from binaries?

Abhishek8857, 2025-02-17:
I get this message: 

root@abhishek:/overlay_ws# apt list --installed | grep geometric-shapes

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

ros-humble-geometric-shapes/jammy,now 2.3.1-1jammy.20241206.150207 amd64 [installed,automatic]

I used the docker image i built yesterday to test this command. I am recieving the said error from today although i havent made any changes to the code

sea-bass, 2025-02-17:
Maybe the Humble sync from Feb 14 has taken effect and you need to do a clean rebuild to avoid version mismatches?
https://discourse.ros.org/t/new-packages-for-humble-hawksbill-2025-02-14/42061

It should be at 2.3.2 now, which is why I think it's worth trying this. Same with MoveIt itself, it should now be 2.5.8.

Abhishek8857, 2025-02-18:
Solved