moveit/moveit2 #3361: Pilz planning times are significantly slower than ROS1

rr-tom-noble, 2025-02-24:
### Description

Planning identical toolpaths in ROS2 using Pilz is significantly slower than ROS1. Timing the Pilz code has identified a couple of significant bottlenecks which appear to relate to `PlanningScene` operations.

### ROS Distro

Humble

### OS and version

Ubuntu 22.04

### Source or binary build?

Source

### If source, which branch?

2.5.8

### Which RMW are you using?

FastRTPS

### Steps to Reproduce

1. Pick a long toolpath (see screenshot below for the one I've been using)
2. Plan the toolpath using GetMotionSequence configured to use Pilz in ROS1 (noetic)
3. Repeat for ROS2 (humble)
4. The planning for ROS2 is much slower

Toolpath:

[image]

### Expected behavior

Planning times on the same toolpath between ROS1 and ROS2 should be comparable.

### Actual behavior

Planning in ROS1 on the given toolpath takes around 0.8 secs. Planning the same toolpath in ROS2 takes around 25.1 secs.

I've stepped through the function calls and added timing statements, which have shown that:
- Around 99% of the time is spent in the GetMotionSequence service, specifically within Pilz
- Around 52% of the time is spent specifically calling `PlanningScene::diff()` [here](https://github.com/moveit/moveit2/blob/bf01f4c9d4516b73095d712da682ad0b74d127b5/moveit_planners/pilz_industrial_motion_planner/src/trajectory_generator.cpp#L357)
- Around 41% of the time appears to be lost after [this](https://github.com/moveit/moveit2/blob/bf01f4c9d4516b73095d712da682ad0b74d127b5/moveit_planners/pilz_industrial_motion_planner/include/pilz_industrial_motion_planner/planning_context_base.h#L124) call, but before control is returned.
   - This was confirmed by timing `TrajectoryGenerator::generate()` both inside and outside
   - This appears to be a direct call, with no other method calls in between
   - The most likely cause I can think of is a slow destructor for a local variable
   - [This](https://github.com/moveit/moveit2/blob/bf01f4c9d4516b73095d712da682ad0b74d127b5/moveit_planners/pilz_industrial_motion_planner/src/trajectory_generator.cpp#L324) could be a likely suspect, as it has a shared ptr to the scene created via `diff()` (I believe the reference count drops to 0 on deletion, as the scene appears to be created and used within this function only)
- The rest of the time appears fairly evenly spread out by comparison. Only these stand out as big bottlenecks

rr-tom-noble, 2025-02-24:
The slow part of `PlanningScene::diff()` appears to be the two `CollisionDetectorAloocator::allocateEnv()` [here](https://github.com/moveit/moveit2/blob/bf01f4c9d4516b73095d712da682ad0b74d127b5/moveit_core/planning_scene/src/planning_scene.cpp#L235-L237)

TSNoble, 2025-02-24:
We're using the default collision detection (FCL I believe). I've noticed [this](https://github.com/moveit/moveit/pull/3601) PR in moveit, which does not appear to be in humble. There is [a commit](https://github.com/moveit/moveit2/commit/1d93035fc4fbc2831916bce26d7b03e15b8a86d5) porting it to more recent versions, but I can't seem to find an associated PR. I'll try applying that fix tomorrow.

TSNoble, 2025-02-24:
Also [this](https://github.com/moveit/moveit/pull/3599) moveit PR. The relevant moveit2 PR appears to be [here](https://github.com/moveit/moveit2/pull/2942)

rr-tom-noble, 2025-02-25:
Just tested these changes, and they don't appear to affect the planning time significantly. Planning to do some more thorough combing through the FCL collision detector to determine which operations are slow.

rr-tom-noble, 2025-02-25:
Further timing confirms that almost all of the 52% bottleneck is spent constructing `CollisionEnv`s [here](https://github.com/moveit/moveit2/blob/bf01f4c9d4516b73095d712da682ad0b74d127b5/moveit_core/collision_detection/include/moveit/collision_detection/collision_detector_allocator.h#L87). A negligible amount of time is spent in the corresponding [CollisionEnvFcl constructor](https://github.com/moveit/moveit2/blob/bf01f4c9d4516b73095d712da682ad0b74d127b5/moveit_core/collision_detection_fcl/src/collision_env_fcl.cpp#L157-L172) and [superconstructor](https://github.com/moveit/moveit2/blob/bf01f4c9d4516b73095d712da682ad0b74d127b5/moveit_core/collision_detection/src/collision_env.cpp#L110-L115), however...

rr-tom-noble, 2025-02-25:
Furthermore, the problem doesn't appear to scale with the sizes of meshes in the world, only the number of points in the path

rr-tom-noble, 2025-02-26:
Turns out there was a simpler answer. We _do_ use an FCL `CollisionEnv`, but it's wrapped in our own collision plugin, which is where the majority of time was being spent. Switched out to pure FCL and, while plans are still a little slower in ROS2, they're much less so now.

closed 2025-02-26 by rr-tom-noble