moveit/moveit2 #3363: fatal error: srdfdom/model.h: No such file or directory when building pkg that includes move_group_interface.h

LUGGGI, 2025-02-26:
### Description

Getting `/opt/ros/humble/include/moveit/robot_model/robot_model.h:42:10: fatal error: srdfdom/model.h: No such file or directory` when compiling packege that includes the `move_group_interface.h`.
Worked bevor i did an `apt upgrade` on 26/02/2025

### ROS Distro

Humble

### OS and version

Ubuntu 22.04

### Source or binary build?

Binary

### Which RMW are you using?

None

### Steps to Reproduce

- install movit with `apt install ros-humble-moveit*`
- `#include <moveit/move_group_interface/move_group_interface.h>`
- colcon build --symlink-install

### Expected behavior

It should compile without error, as it did bevore i did an `apt upgrade` today. The last time bevor that I did an update on 2025-02-04

### Actual behavior

Throws error
`
Starting >>> open_manipulator_x_control
--- stderr: open_manipulator_x_control                             
gmake[2]: *** No rule to make target '/opt/ros/humble/lib/libmoveit_move_group_interface.so.2.5.7', needed by 'open_manipulator_x_control'.  Stop.
gmake[2]: *** Waiting for unfinished jobs....
gmake[2]: *** No rule to make target '/opt/ros/humble/lib/libmoveit_move_group_interface.so.2.5.7', needed by 'libopen_manipulator_x_control_plugin.so'.  Stop.
gmake[2]: *** Waiting for unfinished jobs....
In file included from /opt/ros/humble/include/moveit/robot_state/robot_state.h:40,
                 from /opt/ros/humble/include/moveit/move_group_interface/move_group_interface.h:41,
                 from /home/ros/TurtleBot/ros2_ws/src/open_manipulator_x_control/src/open_manipulator_x_control.cpp:38:
/opt/ros/humble/include/moveit/robot_model/robot_model.h:42:10: fatal error: srdfdom/model.h: No such file or directory
   42 | #include <srdfdom/model.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
`

LUGGGI, 2025-02-27:
For some reason the problem fixed itself. If anyone else is having the same problems, here are the steps I did, I don't know what fixed it, but it started working again after the last step (I also tried in between each step without success):

1. Reinstalling the moveit pkgs
2. going into `/opt/ros/humble/include/moveit/robot_model/robot_model.h` and changing `#include <srdfdom/model.h>` to `#include <srdfdom/srdfdom/model.h>`
3. doing the same thing for other files that threw the same error or an error related to `#include <eigen_stl_containers/eigen_stl_vector_container.h>`

At this point the include errors where gone but i got the error:
`gmake[2]: *** No rule to make target '/opt/ros/humble/lib/libmoveit_move_group_interface.so.2.5.7', needed by 'open_manipulator_x_control'.  Stop.`

4. Then I ran the colcon build command again with ` --event-handlers console_cohesion+ --cmake-args -DCMAKE_VERBOSE_MAKEFILE=ON` (tried this to see more information about the buid)

This build was successful.

Now even the build without the argument succeeds, even when I reset the files I changed in steps 2 and 3

simonleonard, 2025-03-05:
Had the same error after apt upgrade. 

Cleaning the workspace and rebuild solved the problem for me.