moveit/moveit2 #3378: servo twist cmd cause to jump robot previes start state

AltinayGrass, 2025-03-06:
### Description

Please see in attached screen record. There is no error on screen logs.

### ROS Distro

Jazzy

### OS and version

24.04

### Source or binary build?

Binary

### If binary, which release version?

ros-jazzy-moveit-servo/noble,now 2.12.1-1noble.20250202.034934 amd64 [installed]

### If source, which branch?

jazzy

### Which RMW are you using?

FastRTPS

### Steps to Reproduce

ros2 launch sr80_moveit_config sr80_servo_demo.launch 
select command type as twist.

### Expected behavior

Should start current state

### Actual behavior

robot arm Jump to start state and getting servo commands at that point

### Backtrace or Console output

https://github.com/user-attachments/assets/8479416a-7bad-4564-a662-f0cfeec468e5

sea-bass, 2025-03-06:
Should be fixed in https://github.com/moveit/moveit2/pull/3347, which is slated to be out in 2.12.2 on Jazzy.

For now, you can build from source!

sea-bass, 2025-03-08:
BTW seems like Jazzy sync is coming next week!

https://discourse.ros.org/t/preparing-for-jazzy-sync-2025-03-10/42439

closed 2025-03-10 by sea-bass