moveit/moveit2 #3379: prevent to log same warring messages

AltinayGrass, 2025-03-08:
### Is your feature request related to a problem? Please describe.

Since the servo node is opened to subscribe to jog and twist topics, a new warring message line is constantly written to the screen and logs until the command type is entered.

### Describe the solution you'd like

```
void ServoNode::twistCallback(const geometry_msgs::msg::TwistStamped::ConstSharedPtr& msg)
{
  latest_twist_ = *msg;
  if (latest_twist_.twist.linear.x != 0 || latest_twist_.twist.linear.y != 0 || latest_twist_.twist.linear.z != 0)
    new_twist_msg_ = true;
}
```

sea-bass, 2025-03-09:
Interesting! What I think you may want here is to set all the `new_X_message_` variables to `false` right when the command type switches, so adding to this line:

https://github.com/moveit/moveit2/blob/7284d361392259a853ca0fa48448563372d328bb/moveit_ros/moveit_servo/src/servo_node.cpp#L188

Would you be willing to try that, see if it works, and submit a pull request if so?

AltinayGrass, 2025-03-11:
Thank you for your offer, I think I am not yet familiar with the code to the point of being able to do a pull and push. I would prefer to follow the source and add single line changes as needed.