moveit/moveit2 #3380: Noisy velocity command output

thibpn, 2025-03-11:
### Description

I'm trying to command in velocity a robot by publishing delta_twist commands. The output that I have in **velocity_command** is extremely noisy.
Here is an example on the first two axes : 

[image]

To understand where this noise comes from, I installed the source build (humble branch). By modifying it, I identified the Butterworth filter as the source of the noise. When I delete it, I get clean output.

From what I understand, the filter is applied on the joint_states_position + delta_theta. Then, to compute the velocity command : 
  `velocity_command = (filtered(joint_states_position + delta_theta) - joint_states_position) / publish_period`

So if my joint_state_position is noisy, my velocity_command will be noisy, which would not have been the case if we used delta_theta to compute velocity commands.

I wish to stay with the binary build. Is there a way to not use any smoothing plugin in Moveit Servo ?

### ROS Distro

Humble

### OS and version

Ubuntu 22.04

### Source or binary build?

Binary

### If binary, which release version?

2.5.8-1jammy.20250210.235402

### Which RMW are you using?

CycloneDDS

### Steps to Reproduce

I send delta_twist commands. It need to be slow movement so that `delta_theta ~ noise on joint_state`.
Then plot velocity_command.

### Expected behavior

**velocity_command** output should not be noisy.

### Actual behavior

**velocity_command** output is noisy.

AndyZe, 2025-03-11:
It's completely optional to have a smoothing plugin as you can see here:

https://github.com/moveit/moveit2/blob/7284d361392259a853ca0fa48448563372d328bb/moveit_ros/moveit_servo/src/servo.cpp#L166

Or, you can try one of the other smoothing options described here:

https://moveit.picknik.ai/main/doc/examples/realtime_servo/realtime_servo_tutorial.html#signal-smoothing

lparadasanmartin, 2025-03-12:
Hi @AndyZe , i'm very much interested into knowing how to deactivate the smoother filter ! 

I tried commenting the line `# smoothing_filter_plugin_name: "online_signal_smoothing::ButterworthFilterPlugin"` on my `servo.yaml` file but the filter seems to still apply.

How can I deactivate the smoother for sure ? Thanks a lot for your answer & help.

sea-bass, 2025-03-12:
I think since you're using the Humble version, disabling smoothing (or any of the other plugin alternatives) is not available:

https://github.com/moveit/moveit2/blob/954a027f73409676aba34a87fbf23822f6842a6a/moveit_ros/moveit_servo/src/servo_calcs.cpp#L160

If you upgrade to the Jazzy or Rolling (main) version, there is a parameter to simply disable smoothing. For example:

https://github.com/moveit/moveit2/blob/7284d361392259a853ca0fa48448563372d328bb/moveit_ros/moveit_servo/config/panda_simulated_config.yaml#L30