moveit/moveit2 #3404: Dual arm planning error with IKConstraintSampler in Humble, backport feature from Jazzy

gennscar, 2025-03-27:
### Description

Dual arm planning works when using the Rviz MoveIt plugin but fails via the C++ API. 
The error I get is the following one: _**"IKConstraintSampler received dirty robot state, but valid transforms are required. Failing."**_

I see that the issue [Dual arm planning error with IKConstraintSampler #1145
](https://github.com/moveit/moveit2/issues/1145) has been already closed and merged for Moveit2 Jazzy. I suggest backporting it for humble.

This is my code:

```
move_group_interfaces_["both_lite6"]->clearPoseTargets();
move_group_interfaces_["both_lite6"]->setStartStateToCurrentState();

geometry_msgs::msg::Pose L_lite6_pose = move_group_interfaces_["L_lite6"]->getCurrentPose("L_link_eef").pose;
L_lite6_pose.position.x -= 0.05;
L_lite6_pose.position.y -= 0.05;
move_group_interfaces_["both_lite6"]->setPoseTarget(L_lite6_pose, "L_link_eef");

geometry_msgs::msg::Pose R_lite6_pose = move_group_interfaces_["R_lite6"]->getCurrentPose("R_link_eef").pose;
R_lite6_pose.position.x -= 0.05;
R_lite6_pose.position.y -= 0.05;
move_group_interfaces_["both_lite6"]->setPoseTarget(R_lite6_pose, "R_link_eef");

moveit::planning_interface::MoveGroupInterface::Plan pose_plan;
auto planning_result = move_group_interfaces_["both_lite6"]->plan(pose_plan);
```

### ROS Distro

Humble

### OS and version

Ubuntu 22.04

### Source or binary build?

Source

### If source, which branch?

Humble

### Which RMW are you using?

CycloneDDS

### Steps to Reproduce

In my case I am running a simulation with two Ufactory Lite6 robots, but the same behavior can be reproduced following the following test files:
[dual_arm_robot_state_update.py](https://github.com/moveit/moveit2/blob/4a369b6ee1031a06a2e5273c20f650434cb1fade/moveit_ros/planning_interface/test/dual_arm_robot_state_update.py)

[dual_arm_robot_state_update.test](https://github.com/moveit/moveit2/blob/4a369b6ee1031a06a2e5273c20f650434cb1fade/moveit_ros/planning_interface/test/dual_arm_robot_state_update.test)

### Expected behavior

Planning shall return success or fail if no solution has been found.

### Actual behavior

Planning returns a IKConstraintSampler error.

sea-bass, 2025-03-28:
What is the Jazzy/Rolling commit that fixed the problem and should be backported to Humble? If you have the link I can mark it for backport.

gennscar, 2025-03-28:
@sea-bass [Commit 4a369b6](https://github.com/moveit/moveit2/commit/4a369b6ee1031a06a2e5273c20f650434cb1fade) is the one solving the issue. 
In particular I modified UnionConstraintSampler::sample in moveit_core/constraint_samplers/src/union_constraint_sampler.cpp as shown in the commit and everything works.

In jazzy I see the following in June 2022:
[image]

But in humble I do not see this commit:
[image]

gennscar, 2025-04-02:
Okay perfect! I'll test it and open a pull request ASAP, thanks!

closed 2025-07-10 by github-actions[bot]

TDC-TangChaoBanLi, 2025-11-26:
> Okay perfect! I'll test it and open a pull request ASAP, thanks!

Hi， I have also encountered this issue. Do you have any plans to open a pull request recently to fix this problem?