moveit/moveit2 #3405: Change hard-coded timeout in controller manager plugin to a ROS parameter

sea-bass, 2025-03-30:
### Is your feature request related to a problem? Please describe.

As documented in https://github.com/moveit/moveit2/pull/3382#pullrequestreview-2688278559, the `SERVICE_CALL_TIMEOUT` constant can be instead changed to a ROS parameter so users can tune the value for different applications.

### Describe the solution you'd like

Changing the constant to a parameter.

ashwinsnambiar, 2025-04-03:
Hi, can I start working on this?

sea-bass, 2025-04-03:
No need to ask for permission when it comes to OSS -- if you put up a PR, it will be reviewed and appreciated!

DhananjayKhandelwal, 2025-04-06:
Hello @sea-bass! Can you please guide me to a repository that includes a python launch file that uses the Ros2ControlManager plugin, so I can test the changes? Or something else that I can do, any tutorial to use the plugin, maybe. I've been unable to find much. Thanks!

sea-bass, 2025-04-06:
I'm not very familiar with this code, so I may not be the right person to ask in this case.

ashwinsnambiar, 2025-04-07:
I tested the implementation using the https://github.com/moveit/moveit_resources/tree/ros2/dual_arm_panda_moveit_config example moveit configuration by trying out the following values for the parameter in the `moveit_controllers.yaml`:

`controller_service_call_timeout: 0.00001` : Controllers did not connect
`controller_service_call_timeout: 1.0` : Controllers connected and path planning could be executed

Hope this is fine?

ashwinsnambiar, 2025-04-07:
I have made few additions in the moveit2_tutorial repo too. Please let me know if it is necessary. If so, any corrections to be made?
https://github.com/moveit/moveit2_tutorials/compare/main...ashwinsnambiar:moveit2_tutorials:main

mikeferguson, 2025-04-15:
Closed by #3419 (back ports are still in process, but I think it's safe to close this)

closed 2025-04-15 by mikeferguson