moveit/moveit2 #3411: Moveit2 Servo Always giving "Very close to a singularity, emergency stop"

CanavarB, 2025-04-02:
### Description

Hi, I have a 4DOF robot which I designed. I want use realtime servo on this robot. I launched moveit servo with my config. For the first try I am publishing some twist commands for servo node but every time it says "Very close to a singularity, emergency stop". I can control my robot with "normal" moveit (approx IK solition checked). I changed many times servo config yaml for singularity thresholds.

### ROS Distro

Jazzy

### OS and version

24.04.2

### Source or binary build?

Source

### If binary, which release version?

From ROS repo

### Which RMW are you using?

None

### Steps to Reproduce

Normal Ubuntu Desktop.

### Expected behavior

Every time I get "Very close to a singularity, emergency stop" warning.

### Actual behavior

.

### Backtrace or Console output

[servo_node-10] [WARN] [1743601979.331145654] [servo_node]: Command type has not been set, cannot accept input
[servo_node-10] [WARN] [1743601979.431078677] [servo_node]: Command type has not been set, cannot accept input
[servo_node-10] [WARN] [1743601979.531184001] [servo_node]: Command type has not been set, cannot accept input
[servo_node-10] [WARN] [1743601979.631153174] [servo_node]: Command type has not been set, cannot accept input
[servo_node-10] [WARN] [1743601979.731071291] [servo_node]: Command type has not been set, cannot accept input
[servo_node-10] [WARN] [1743601979.831163947] [servo_node]: Command type has not been set, cannot accept input
[servo_node-10] [WARN] [1743601979.931170629] [servo_node]: Command type has not been set, cannot accept input
[servo_node-10] [WARN] [1743601980.031155456] [servo_node]: Command type has not been set, cannot accept input
[servo_node-10] [WARN] [1743601980.131110808] [servo_node]: Command type has not been set, cannot accept input
[servo_node-10] [WARN] [1743601980.231149954] [servo_node]: Command type has not been set, cannot accept input
[servo_node-10] [WARN] [1743601980.331077102] [servo_node]: Command type has not been set, cannot accept input
[servo_node-10] [WARN] [1743601980.431096715] [servo_node]: Command type has not been set, cannot accept input
[servo_node-10] [WARN] [1743601980.531079188] [servo_node]: Command type has not been set, cannot accept input
[servo_node-10] [WARN] [1743601986.550210444] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.564639587] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.571686492] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.578887354] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.588086328] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.600429969] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.609295698] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.619065678] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.628548131] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.638265059] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.648080424] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.658230357] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.669463791] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.678033992] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.688745632] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.698252258] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.708473796] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.719792966] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.728455970] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.738572917] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.749285112] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.758745102] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.769359863] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.778573555] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.789049563] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.798646259] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.808099833] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.818434643] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.828273458] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.838899505] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.850128470] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.859413975] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.868949431] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.879273861] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.890723136] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.899228164] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.909407502] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.919396731] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.929425534] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.939391215] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.949519807] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.958411113] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.969162613] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.980250421] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601986.988959270] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601987.000325632] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601987.008715786] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601987.019002211] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601987.029663035] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop
[servo_node-10] [WARN] [1743601987.038640593] [servo_node.servo_node.moveit.ros.servo]: Very close to a singularity, emergency stop

sea-bass, 2025-04-04:
I assume that for a 4 DOF arm, which has less than 6 degrees of freedom, the hacky check in MoveIt Servo to pause near singularities due to Jacobian condition number is firing almost no matter what.

I'd say that MoveIt Servo's twist/pose command mode probably will not work well for arms with less than 6 DOF, and you may be better suited implementing your own mechanism to track these as best as possible with the reduced DOFs.

If you are simply jogging this robot's joints (not in Twist mode), you'll probably be fine though.

CanavarB, 2025-04-05:
I saw [4dof](https://github.com/ROBOTIS-GIT/open_manipulator/issues/192) people are adding 2 extra virtual joints to reduce this problem. What is your opinion about this?

sea-bass, 2025-04-05:
> I saw [4dof](https://github.com/ROBOTIS-GIT/open_manipulator/issues/192) people are adding 2 extra virtual joints to reduce this problem. What is your opinion about this?

Hmm, seems like a big workaround.

Servo has an option to use the robot model's Jacobian pseudoinverse instead of an IK solver, but it doesn't have a damping term for the Damped Least Squares / Levenberg-Marquardt approach, which would help a lot for numerical stability in these conditions.

See [this paper](https://www.google.com/url?sa=t&source=web&rct=j&opi=89978449&url=https://www.cs.cmu.edu/~15464-s13/lectures/lecture6/iksurvey.pdf&ved=2ahUKEwiTzsbUhsGMAxX-hIkEHc-hEhsQFnoECEgQAQ&usg=AOvVaw3eWMUb1kuE997WkypZ2LK3)

I feel like this would be what I would try in your situation, but it would require some source code modifications.

AndyZe, 2025-04-06:
The real solution would be to drop 2 rows (or is it columns?) from the Jacobian when you're using the pseudoinverse method. Damped least squares isn't going to help with this issue.

We used to have the capability to do that very easily, which was discussed in this issue:  https://github.com/moveit/moveit_msgs/issues/185

In fact, let's move this conversation over there. You aren't the only one, @CanavarB !

Maybe you can re-implement the feature mentioned there.

closed 2025-04-06 by AndyZe

sea-bass, 2025-04-06:
Not sure how dropping 2 rows from the Jacobian will help, as the rows correspond to actual Cartesian degrees of freedom (specifically XYZ translational and rotational velocities), which is very much configuration dependent. Unless your robot is e.g. on a linear gantry where you know you never have motion along a specific axis, that probably won't work.

However, the discussion in that thread around allowing approximate IK solutions is definitely worthwhile.

Ideally one can produce a joint velocity vector that "best" tracks the provided Cartesian command. DLS can at least help with numerical stability in the rank-deficient case, but approximate IK would indeed be better.

sea-bass, 2025-04-06:
Yeah, thinking about this some more I guess there probably isn't a generic way to handle this with just a Jacobian pseudoinverse solver, so letting the user say that e.g. they don't care about pitch/roll dimensions makes sense. Thanks for the explanation!

You still might get back into these warnings if you happen to null out the "wrong" dimensions, or otherwise get yourself into a singularity in this reduced space (e.g., the 4DOF arm is fully outstretched).

To that end, I still wonder if damped least-squares could still help bypass the check that barfs near singularities (at least when using the Jacobian solver), but it's definitely an orthogonal problem. It's task dependent whether one would prefer to slow down/stop altogether vs. use the "fudged" solution from the damped pseudoinverse but at least keep the arm moving. Maybe having this be a configurable parameter could be useful as an additional feature, on top of reinstating the feature in the other thread?

AndyZe, 2025-04-06:
All of those features sound cool

CanavarB, 2025-04-12:
I solved with adding 2 extra axis joint. I didn't test the servo yet but in "normal" moveit my ik solver (bio_ik) is solving positions very well. My robot have Z X X Z revolute joints and a gripper. I added 2 joints (and links) to the grippers edge and the ik chain. One of them is X axis and the other one is Y axis.

CanavarB, 2025-04-15:
I also tested it on Servo. Twist commands work with very large lower_singularity_threshold.