moveit/moveit2 #3416: Some Strong opinion on moveit2

maxspahn, 2025-04-03:
### Description

Hi everyone, 

thanks for all the developement in moveit and I am sure many people use it with joy. 
However, i have been trying to set it up for a new robot in jazzy for 2 weeks now and there is no way that this package is usable. I will list some issues from a user's perspective. I am sure it is all the user's fault, so I don't expect any reply and this issue should rather be seen as a warning to people tempted to use this piece of software.

1. This is a fairly central component of ROS2. Why is it necessary to install moveit2 from source to walk through the tutorials, see https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html? This is prone to issues and frustration in the build process. This package might be the first contact point for some people with ROS. 
2. From the documentation it is still unclear to me, how I am supposed to interact with moveit from code. The rviz example is all cool and nice, but not something useful for coding an actual project, regardless of whether it is for industral or academic purposes. There is the API and there is the action-client way. The action client version is hardly documented, but was working quite ok in ROS1. Till that day, I have not been able to construct tasks with the list of constraints, a simple pose constraint does not seem to exotic. Then, there is the API (I used the python-bindings version of it). There is so much confusion around here as well. In order to create an MoveItPy instance, you need a moveit config. Stupid me trying to use the same config as in the launch file I guess. The very first code block on https://moveit.picknik.ai/main/doc/examples/motion_planning_python_api/motion_planning_python_api_tutorial.html, where should i put this? Which node requires it? After finding out that for some reason you need to load a moveit_cpp.yaml file for the Moveitpy, i kind of got it working, but I have no clue which planner configurations are actually used, as constantly some defaults are being loaded.
3. The things about nested launch files. When running the setup_assistant, you get a nice looking package with a bunch of launch files. But god forbid you want to change something about the launch files. They all refer back to the moveit_config_utils launch generation. This package is a pure python package where you do not have any access rights. Actual valuable/editable launch files can be better generated using the moveit_resources, for example https://github.com/moveit/moveit_resources/blob/ros2/panda_moveit_config/launch/demo.launch.py. Maybe that is just me being nostalgic about the xml format for launch files.
4. What about casting of integers? When I generated my moveit_config package for my robot, a standard of-the-shelf industrial robot (no self built or anything), I ended up with a joint_limits.yaml without acceleration limits, which makes moveit very unhappy. Then I set some arbitrary acceleration limits to test whether that was the problem. It was indeed, but you have to make very sure that you use decimal values, because moveit does not work with integer values there. Specifically setting the acceleration limit to `1.0` is fine, `1` is not. This issues happened in multiple locations, but this is the one I remember best, because it cost me half a day to figure that out.
5. I am honestly surprised how this piece of software, in my view central to ROS as motion planning is kind of important for robots, is being used so widely. I am pretty sure it is just a user's mistake but the fact that there are currently 200+ open issues make me think that i am not the only one enjoying it.

Cheers everyone, 
enjoy the summer.
@maxspahn

### ROS Distro

Humble

### OS and version

Ubuntu 22.04

### Source or binary build?

Source

### Which RMW are you using?

None

### Steps to Reproduce

Just follow the tutorial.

### Expected behavior

Working examples that i can edit.

### Actual behavior

Not working tutorial

sea-bass, 2025-04-04:
There are two types of software: those that people complain about, and those that nobody uses.

PyTorch is an extremely popular and widely adopted project, and it has over 5000 issues: https://github.com/pytorch/pytorch

That said, you're right. MoveIt 2 is quite challenging to use, and not maintained actively enough. So this project needs help from the community to either address some of these issues, or to make better tools altogether!

mikeferguson, 2025-04-15:
> From the documentation it is still unclear to me, how I am supposed to interact with moveit from code.

Did you try the next tutorial in the sequence - [Planning Around Objects](https://moveit.picknik.ai/main/doc/tutorials/planning_around_objects/planning_around_objects.html) - this uses the move_group interface to control the arm.

> Then, there is the API (I used the python-bindings version of it).

Are you actually using Humble? If so, there is no Python API - it was added after Humble.

> What about casting of integers? When I generated my moveit_config package for my robot, a standard of-the-shelf industrial robot (no self built or anything), I ended up with a joint_limits.yaml [...]

This is actually a ROS 2 parameter thing - they are strongly typed and so an integer cannot be cast to a double (and we really can't control that from MoveIt - it's a function of the whole ROS 2 parameter interface).

> I am honestly surprised how this piece of software, in my view central to ROS as motion planning is kind of important for robots, is being used so widely. I am pretty sure it is just a user's [...]

Counterpoint - if nobody was using it, there would be no issues. This is a complex piece of software, supporting many features across a very large number of robot configurations, so there are going to be lots of edge cases and competing needs which lead to a number of issues (I think there's also about 50-75 issues open for components, and especially tests, that haven't yet been ported to ROS 2).

I think most long term users eventually find that while there is a steep learning curve, it's still quite a bit faster to learn to use the existing software than it would be to write their own from scratch.

I'm going to close this issue, because I don't see any particularly actionable items that aren't already ticketed elsewhere ("make the docs better" is too broad to actually declare "hey this issue has been completed"). I'd also suggest that you check out robotics.stackexchange.com which can be a more effective way to get assistance on individual issues as you encounter them (especially for ones that are more use case specific)

closed 2025-04-15 by mikeferguson