moveit/moveit2 #3420: Filter robot arm from point cloud in Moveit2

marklouis93, 2025-04-07:
### Is your feature request related to a problem? Please describe.

**Version**
Humble

**OS**
Ubuntu22.04

**Install**
moveit2 from source on branch Humble

**Request**

I am trying to make a robot arm to avoid object collision. This collision object is obtained by creating a an octomap from a point cloud, as described on [Moveit2 Perception Pipeline Tutorial](https://moveit.picknik.ai/main/doc/examples/perception_pipeline/perception_pipeline_tutorial.html). 

On the real life case, we want to live stream the environment to generate the octomap, where the robot arm is still there. The robot arm is still in the octomap creation, making it described as an obstacle later on. I want to remove/filter the robot arm from the octomap, so only the remaining object will be regarded as the obstacle.

I have been looking for the older Moveit 1 repositories, and found the application to do a mesh filter in [here](https://moveit.github.io/moveit_tutorials/doc/mesh_filter/mesh_filter_tutorial.html). But it is in ROS1 Noetic.

Is there any package that do the same with ROS2 version?

### Describe the solution you'd like

A desired solution would be similar to this is [ROS1 mesh filter](https://moveit.github.io/moveit_tutorials/doc/mesh_filter/mesh_filter_tutorial.html), but the implementation is in ROS2.

Ayush1285, 2025-04-07:
Hi, I had the same issue a few weeks ago. Feel free to check this issue filed by me, #3285. There are two parameters, "padding_offset"  and "padding_scale,"  which can be tuned to inflate your robot's mesh.

marklouis93, 2025-04-14:
Thank you for the explanation and pointers, cheers.

closed 2025-04-14 by marklouis93