moveit/moveit2 #3431: MoveIt2 trajectory execution fails in RViz when spawner could not contact controller service

Zhang-Qingyang, 2025-04-12:
### Description

Hi MoveIt2 community,
At first, I'm not a English native speaker, please forgive that the following statements may contain some syntax issues or spelling mistakes.

## Problem Descriptions
I encountered two problems when debugging with a ROS2 working space using MoveIt2.
1.**[main problem]** When executing trajectory in RViz2, the terminal logs shows failure. The logs are as follow:
```
[move_group-3] [INFO] [1744434279.216024205] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Received goal request
[move_group-3] [INFO] [1744434279.216204353] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution request received
[move_group-3] [INFO] [1744434279.216249438] [moveit.plugins.moveit_simple_controller_manager]: Returned 1 controllers in list
[move_group-3] [INFO] [1744434279.216276809] [moveit.plugins.moveit_simple_controller_manager]: Returned 1 controllers in list
[move_group-3] [INFO] [1744434279.216416021] [moveit_ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[rviz2-1] [INFO] [1744434279.216305724] [move_group_interface]: Execute request accepted
[move_group-3] [INFO] [1744434280.216509370] [moveit_ros.current_state_monitor]: Didn't receive robot state (joint angles) with recent timestamp within 1.000000 seconds. Requested time 1744434279.216431, but latest received state has time 0.000000.
[move_group-3] Check clock synchronization if your are running ROS across multiple machines!
[move_group-3] [WARN] [1744434280.216547592] [moveit_ros.trajectory_execution_manager]: Failed to validate trajectory: couldn't receive full current joint state within 1s
[move_group-3] [INFO] [1744434280.216633233] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution completed: ABORTED
[rviz2-1] [INFO] [1744434280.217049515] [move_group_interface]: Execute request aborted
[rviz2-1] [ERROR] [1744434280.217246535] [move_group_interface]: MoveGroupInterface::execute() failed or timeout reached
```
2.**[derived problem]** Even if it's unrelevent to MoveIt2，I think it may be helpful to solve the issue. After the spawner node starts, the terminal repeatedly print following message:
```
[spawner-6] [WARN] [1744434281.014963849] [spawner_engineer_group_controller]: Could not contact service /controller_manager/list_controllers
[spawner-5] [WARN] [1744434281.014964450] [spawner_joint_state_broadcaster]: Could not contact service /controller_manager/list_controllers
[spawner-5] [INFO] [1744434281.015592450] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[spawner-6] [INFO] [1744434281.015592430] [spawner_engineer_group_controller]: waiting for service /controller_manager/list_controllers to become available..
```

[image]

## Context information and Attampted Fixs
**My working space was named `controller`**（there may be some ambiguilty，I'm means that the FOLDER NAME of working space is `controller`. I‘m not sure whether it matter, but it may cause some potential name conflicts）
There is only one planning group inside the working space，which name is **`engineer_group`**，corresponding MoveIt configuration package is `engineer_moveit_config`.

The following attampts was conducted from other terminal when the relevant nodes was executing(ROS2 and MoveIt2 environments has been sourced)
```
poou@poou-22:~$ ros2 node list
/controller_manager
/interactive_marker_display_101900220924864
/kine_node
/move_group
/move_group_private_96089974971808
/moveit_simple_controller_manager
/robot_state_publisher
/rviz2
/rviz2_private_134418220025056
/serial_node
/spawner_engineer_group_controller
/spawner_joint_state_broadcaster
/transform_listener_impl_5764b2326860
/transform_listener_impl_5cad7e3b1050
/transform_listener_impl_5cad7e4e7c60
/transform_listener_impl_7a40ad3d5bd0
```
The logs shows that controller_manager and robot_state_publisher are all executing，which indicates the cause isn't the node has quit。
```
poou@poou-22:~$ ros2 node info /controller_manager
/controller_manager
  Subscribers:
    /controller_manager/robot_description: std_msgs/msg/String
    /parameter_events: rcl_interfaces/msg/ParameterEvent
  Publishers:
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
  Service Servers:
    /controller_manager/describe_parameters: rcl_interfaces/srv/DescribeParameters
    /controller_manager/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
    /controller_manager/get_parameters: rcl_interfaces/srv/GetParameters
[... 139 lines cut ...]

### If source, which branch?

main

### Which RMW are you using?

CycloneDDS

### Steps to Reproduce

Clone the whole working folder from this repo：https://github.com/Zhang-Qingyang/controller/tree/master
Type following in the terminal in the root of working space folder.
```
source install/setup.bash
ros2 launch controller_bringup run.launch.py
```

### Expected behavior

I expected RViz will correctly load all controllers, and execute the plan.

### Actual behavior

When executing trajectory in RViz2, the terminal logs shows failure.

closed 2025-04-18 by Zhang-Qingyang

EnderMandS, 2025-06-20:
I am facing the same problem
``` 
Could not contact service /controller_manager/list_controllers
waiting for service /controller_manager/list_controllers to become available..
```
How did you resolve the issue?

Zhang-Qingyang, 2025-06-20:
> I am facing the same problem
> 
> ```
> Could not contact service /controller_manager/list_controllers
> waiting for service /controller_manager/list_controllers to become available..
> ```
> 
> How did you resolve the issue?

I don't know how exactly your configuration are.If you can provide more information, I'm happy to help you.