moveit/moveit2 #3434: Fix joint trajectory controller duplicate key generation in setup assistant properly

sea-bass, 2025-04-13:
### Description

See https://github.com/moveit/moveit2/pull/3316#issuecomment-2798165248 for more information.

The original fix (#2959) fixed key duplication in the event that the `action_ns` and `default` keys were specified in the UI, but seem to break the workflow if those text boxes are left blank.

There needs to be some if-condition (and hopefully tests) added to properly address both cases.

### ROS Distro

Rolling

### OS and version

Ubuntu 24.p4

### Source or binary build?

Source

### Which RMW are you using?

None

### Steps to Reproduce

no u

### Expected behavior

Should work under both conditions outlined above.

### Actual behavior

Works only if the values are set in the UI, but not when blank.

zacharyyamaoka, 2025-04-13:
Some further testing I did on this:

If you just press "Auto Add FollowJointTrajectory Controllers" and continue on in set up assitant, than the generated .yaml is missing an action_ns, and won't work out of the box

[image]

```
moveit_controller_manager: moveit_simple_controller_manager/MoveItSimpleControllerManager

moveit_simple_controller_manager:
  controller_names:
    - arm_controller
    - eef_controller

  arm_controller:
    type: FollowJointTrajectory
    joints:
      - right_joint_1
      - right_joint_2
      - right_joint_3
      - right_joint_4
      - right_joint_5
      - right_joint_6
  eef_controller:
    type: FollowJointTrajectory
    joints:
      - right_joint_6
  ```

If you however press the "edit selected" button, default values will auto populate:

[image]

```
moveit_controller_manager: moveit_simple_controller_manager/MoveItSimpleControllerManager

moveit_simple_controller_manager:
  controller_names:
    - arm_controller
    - eef_controller

  arm_controller:
    type: FollowJointTrajectory
    joints:
      - right_joint_1
      - right_joint_2
      - right_joint_3
      - right_joint_4
      - right_joint_5
      - right_joint_6
    action_ns: follow_joint_trajectory
    default: true
  eef_controller:
    type: FollowJointTrajectory
    joints:
      - right_joint_6
 ```

You can see the arm_controller now gets an action_ns, but the eef_controller stil doesn't as I didn't select "edit selected" for the eef_controller.

I believe that an action_ns should be added even if the user doesn't select "edit selected", to avoid frusteration with it not working out of the box

sea-bass, 2025-04-13:
Thanks for the extra context!

Yes, it should be a partial revert of the original PR, but the blank/default fields should only be added if the users do not fill in the respective text boxes.

yonteels, 2026-05-01:
Hey,  I also created a solution for this