moveit/moveit2 #3439: Moveit Setup Assistant Crashes when uploading a URDF

romillyn1, 2025-04-14:
### Description

Whenever the URDF/COLLADA box has a character entered into it or you browse and select a file the program freezes and crashes. Have tried installing moveit via binary and source, issues persists.

Have tried multiple URDFs, and even passing the file path via CLI to no avail

No error messages displayed in console other than this warning:

[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [moveit_setup_assistant-1]: process started with pid [66804]
[moveit_setup_assistant-1] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.

### ROS Distro

Humble

### OS and version

Ubuntu 22.04

### Source or binary build?

Binary

### If binary, which release version?

2.5.8-1jammy.20250326.024334

### Which RMW are you using?

None

### Steps to Reproduce

Install moveit2
Run - ros2 launch moveit_setup_assistant setup_assistant.launch.py
Select Browse button
Navigate to URDF

### Expected behavior

Expected to select the file path and enable user to click load file

### Actual behavior

Application freezes indefinetly

nbbrooks, 2025-05-21:
Try checking out https://github.com/moveit/panda_moveit_config/tree/rolling-devel and opening it and see if the Browse button still hangs. I believe MSA requires a package.xml to exist in the folder or parent folder (so that it can make updates to the package.xml as you make modifications via MSA). Obviously, more graceful error handling would be great (and a pull request is welcome!)

nbbrooks, 2025-12-10:
Did you check if you have a package.xml in the folder or parent folder per https://github.com/moveit/moveit2/issues/3439#issuecomment-2898266865 ?

dkworks108, 2026-11-01:
I encountered the same issue while using MoveIt Setup Assistant with ROS 2 Humble on Ubuntu 22.04.

**My environment:**
- Ubuntu 22.04
- ROS 2 Humble
- MoveIt 2 (binary installation)
- Custom 3-DOF robotic arm URDF

**Root cause and solution:**

In my case, the Setup Assistant crashed when I selected my URDF because the project directory did not contain a `package.xml`.

I resolved this by:

1. Creating a valid `package.xml` in the robot description package.
2. Creating a `CMakeLists.txt` to define the ROS 2 package and install its URDF, configuration, and launch directories.
3. Building the package using `colcon build`.
4. Sourcing the ROS 2 workspace using `source install/setup.bash`.
5. Verifying package discovery using `ros2 pkg prefix arm_3dof_description`.

After these steps, MoveIt Setup Assistant successfully loaded my URDF.

**My suggestion:**

It would be helpful if MoveIt Setup Assistant checked whether a valid `package.xml` exists when a user selects a URDF.

If the package manifest is missing, instead of allowing the application to crash or close unexpectedly, it could display an informative error message explaining:
- A ROS 2 package manifest is required.
- Where the `package.xml` should be located.
- How users can create or configure the required package.

An optional feature to generate a basic `package.xml` template could also make the initial setup easier for new ROS 2 users.

This would improve the user experience and make the Setup Assistant's failure easier to diagnose.

Thank you to the maintainers and contributors for the guidance. This issue helped me understand ROS 2 package discovery and MoveIt configuration.