moveit/moveit2 #3440: Regression on ROS 2 Buildfarm for Rolling

Yadunund, 2025-04-14:
### Description

`moveit 2.13.0` is failing to build on the ROS 2 buildfarm for Rolling.
 

### ROS Distro

Rolling

### OS and version

24.04

### Source or binary build?

Binary

### If binary, which release version?

2.13.0

### Which RMW are you using?

None

### Steps to Reproduce

See https://build.ros2.org/view/Rbin_uN64/job/Rbin_uN64__moveit_setup_framework__ubuntu_noble_amd64__binary/140/consoleFull

### Expected behavior

Build succeeds. 

### Actual behavior

```
23:15:40 CMake Error at CMakeLists.txt:9 (find_package):
23:15:40   By not providing "Findament_cmake_ros.cmake" in CMAKE_MODULE_PATH this
23:15:40   project has asked CMake to find a package configuration file provided by
23:15:40   "ament_cmake_ros", but CMake did not find one.
23:15:40 
23:15:40   Could not find a package configuration file provided by "ament_cmake_ros"
23:15:40   with any of the following names:
23:15:40 
23:15:40     ament_cmake_rosConfig.cmake
23:15:40     ament_cmake_ros-config.cmake
23:15:40 
23:15:40   Add the installation prefix of "ament_cmake_ros" to CMAKE_PREFIX_PATH or
23:15:40   set "ament_cmake_ros_DIR" to a directory containing one of the above files.
23:15:40   If "ament_cmake_ros" provides a separate development package or SDK, be
23:15:40   sure it has been installed.
```

Yadunund, 2025-04-14:
The offending package was likely linking against `ament_cmake_ros` transitively but we'll need to be explicit after https://github.com/ros/rosdistro/pull/44971

sea-bass, 2025-04-14:
Thanks for the heads up, solving in #3441

closed 2025-04-15 by sjahr

Yadunund, 2025-04-15:
@sea-bass thanks for the quick fix! Did you also bloom a new release?  That will be needed to fix binary builds on the buildfarm.

sea-bass, 2025-04-15:
> [@sea-bass](https://github.com/sea-bass) thanks for the quick fix! Did you also bloom a new release? That will be needed to fix binary builds on the buildfarm.

I will bloom releases later tonight!

sea-bass, 2025-04-16:
Ok @Yadunund rosdistro PR is here -- thanks again!
https://github.com/ros/rosdistro/pull/45194

PR #3441 (2025-04-15): Use ament_cmake_ros build tool in moveit_setup_assistant packages
### Description

Something changed in rolling apparently, so buildfarm is not happy.

Closes #3440 

### Checklist
- [ ] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [ ] Extend the tutorials / documentation [reference](http://moveit.ros.org/documentation/contributing/)
- [ ] Document API changes relevant to the user in the [MIGRATION.md](https://github.com/moveit/moveit2/blob/main/MIGRATION.md) notes
- [ ] Create tests, which fail without this PR [refere [... cut]