moveit/moveit2 #3454: Include the moveit_py into Humble binary release?

lesurJ, 2025-04-24:
### Is your feature request related to a problem? Please describe.

The python bindings for moveit (moveit_py) are not included in the binary release for ROS2 Humble. A workaround is to build the main branch that contains the moveit_py package directly. However, building and updating this package from source is not as practical as updating a binary release.

### Describe the solution you'd like

It would be nice to have the python bindings released directly in the binary release for Humble. And since the source build of the main branch is working fine on ROS2 humble, the binary build for moveit_py should be easy to set up.

sea-bass, 2025-04-24:
This should be doable since it's a new package so it shouldn't mess with ABI stability.

The big challenge will be that you likely won't be able to just copy-paste the package from `main` since there have been some changes between the `humble` branch being cut and the `moveit_py` package being developed. So some care will have to be taken.

Anyways, if someone wants to work on this and gets it up and running, it would be a good contribution!

SamueleSandrini, 2025-05-29:
Hi @sea-bass, could you please review my pull request? I'm having a small problem running `colcon tests`.
Thank you very much! 😉