moveit/moveit2 #3464: Move_group status

Shobinvinod, 2025-05-06:
In ROS 1, it was possible to monitor the status of the move_group using the /move_group/status topic, but in MoveIt 2 with ROS 2, this topic is no longer available.

I am using moveit_py to control a robot, where I've created a custom action server for controlling the robot’s movements. I am aware that the /move_action topic provides feedback with respect to the goals you send to the action server, but since I am not using the standard move_group action server, sending a "dummy goal" to /move_action doesn't provide meaningful feedback in my case.

Here’s the situation:

Custom Action Server: I have set up my own action server using moveit_py to control robot movements.

MoveIt 2 Action Server Feedback: Even if I send a dummy goal to the /move_action action server, it doesn’t provide useful feedback related to the custom action I created. The feedback would be specific to the goals sent to the default MoveIt action server, but not to my custom action.

Questions:

How can I monitor real-time feedback or status (e.g., PLANNING, EXECUTING, IDLE) for my custom MoveIt 2 action server that I created using moveit_py?

Is there a way to access feedback directly from the custom action server using ROS 2 terminal commands (similar to how /move_action works for the default server)?

If there is no direct terminal command to achieve this, do I need to implement a custom action client to subscribe to the feedback topic, or is there another method for monitoring my custom action server's feedback?

Any advice or guidance on how to achieve this would be much appreciated!

Danilrivero, 2025-05-13:
By executing

```bash
ros2 topic list --include-hidden-topics
```

You can find topics such as 

`ros2 topic echo /move_action/_action/feedback` that you can subscribe to or read through terminal with an _echo_. Unsure if this is exactly what you are looking for.

Shobinvinod, 2025-05-15:
@Danilrivero  , Thank you for the response. 

If I use "ros2 topic echo /move_action/_action/feedback" , the feedback is specific to the goal that is send to the action /move_action. But in my case I want to know the status of move_group, when I control the robot using the moveit-py package.

closed 2025-05-27 by Shobinvinod