moveit/moveit2 #3472: moveit_servo jitter on hardware

ConradKent, 2025-05-15:
### Description

I'm using a physical Kinova Gen3. I have moveit through RVIZ, the twist controller that comes with Kinova's software, and moveit_servo running on the arm. moveit_servo has significant jitter, which vibrates the table it's bolted to and is probably not great for the robot. I don't have this issue with either of the other control schemes. I'm using the keyboard tutorial script to send commands to moveit_servo, the jitter happens with both twist and joint commands.

Do you have any suggestions for tuning the moveit_servo config file or otherwise smoothing out the jitter?

The publish period is set to 0.002 (500hz) in the servo config file from Kinova, I tried setting it to 0.01 and got similar results.

### ROS Distro

Humble

### OS and version

Ubuntu 22.04

### Source or binary build?

Source

### Which RMW are you using?

None

### Steps to Reproduce

Implement moveit_servo on hardware, move hardware. Unfortunately this is probably hardware specific.

### Expected behavior

Smooth, continuous motion of the Kinova Gen3 arm.

### Actual behavior

Vibrational, step-by-step motion of the Kinova Gen3 arm which vibrates the table it's bolted to.

ConradKent, 2025-05-15:
I realized that in the more recent versions of moveit_servo there is a configuration for adding Ruckig smoothing, I assume this would fix the issue I'm having. There's no config option for smoothing plugins on the Humble branch as far as I know. I can close the issue if this config change would solve my vibration problem.

closed 2025-05-15 by ConradKent

AndyZe, 2025-05-15:
See the other smoothing options here:  `https://github.com/moveit/moveit2/tree/main/moveit_core/online_signal_smoothing`

I think the "acceleration" smoothing plugin in Jazzy is also very good. This is the config for it:

```
use_smoothing: true
smoothing_filter_plugin_name: "online_signal_smoothing::AccelerationLimitedPlugin"
```