moveit/moveit2 #3473: Is there an official docker image with moveit_py and humble?

Frank-Schiro, 2025-05-15:
### Is your feature request related to a problem? Please describe.

Is there an official docker image with moveit2 that contains moveit_py? If not, can we please have one?

I tried pulling a few different images but I don't see it in any, for example here is humble-source, it has moveit_commander which I thought was ROS1 but not moveit_py. 

```bash
docker run -it moveit/moveit2:humble-source /bin/bash

root@776becd35b3b:~/ws_moveit# ls src/moveit2
CODE_OF_CONDUCT.md  MIGRATION.md  moveit            moveit_configs_utils  moveit_planners  moveit_setup_assistant
CONTRIBUTING.md     README.md     moveit2.repos     moveit_core           moveit_plugins
Doxyfile            codecov.yaml  moveit_commander  moveit_experimental   moveit_ros
LICENSE.txt         doc           moveit_common     moveit_kinematics     moveit_runtime
```

### Describe the solution you'd like

A docker image on docker hub that contains humble with moveit_py

mikeferguson, 2025-05-16:
The moveit_py bindings are not in the humble branch (they were done after Humble was released, and never backported), and I think we've now diverged enough that the main branch no longer works on Humble. So this would be a pretty hefty task.

I don't know for sure, but I'm guessing the "moveit_commander" folder is there, but not being built in Humble. For a long time, there were a number of unported packages still sitting in the repo with the CMake basically disabled until they were ported/replaced/removed.

Frank-Schiro, 2025-05-16:
I see. So would you say with Humle one shouldn't really use moveit_commander or moveit_py? I guess just ROS action servers if you are using python?

sea-bass, 2025-05-16:
You actually still can build moveit2 `main` branch on Humble, and this Docker image may give you access to `moveit_py` on Humble... I think.

https://hub.docker.com/layers/moveit/moveit2/main-humble-tutorial-source/images/sha256-62ce5f8ab1de0b2091fa335e099c5de4e7de38bd49927e3d9fd27b06103d7e97