moveit/moveit2 #3475: Interactive Marker does not update pose in RViz (returns to initial position on release)

LucaMIMOUNI, 2025-05-22:
### Description

When using MoveIt2 with a Yaskawa robot in Rviz, the interactive marker displays correctly and can be moved while dragging. However, after releasing the marker, it snaps back to its original position and does not update the robot's target pose. The robot can still be controlled by specifying joint states directly, but the interactive marker does not function as expected.

### Expected Behavior
The interactive marker should remain at the new pose after being released, updating the robot's target pose accordingly.

### Actual Behavior
The marker snaps back to its original position after being released, and does not update the target pose.

### Console/Log Output
```
Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open
```

and further

```
[rviz2-6] [INFO] [1747901732.535339185] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[rviz2-6] [INFO] [1747901732.538334492] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[rviz2-6] [INFO] [1747901732.539494494] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/monitored_planning_scene'
[rviz2-6] [INFO] [1747901732.881019435] [interactive_marker_display_99449856356304]: Connected on namespace: /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic
[rviz2-6] [INFO] [1747901732.900993103] [interactive_marker_display_99449856356304]: Sending request for interactive markers
[rviz2-6] [INFO] [1747901732.903066840] [moveit_ros_visualization.motion_planning_frame]: group yaskawa_arm
[rviz2-6] [INFO] [1747901732.903093509] [moveit_ros_visualization.motion_planning_frame]: Constructing new MoveGroup connection for group 'yaskawa_arm' in namespace ''
```

### System Information
- ROS 2 Distro: Humble
- Ubuntu Version: 22.04
- MoveIt2 Version: (please specify if possible)
- Robot Hardware/Simulation: Yaskawa robot in Gazebo

### Additional Information
- No other errors observed. Robot can be moved by specifying joint states.
- The issue appears related to the warning about class loader namespace collision.

Please let me know if further logs or details are required.

Flamethr0wer, 2025-07-01:
I am having the exact same problem. Did you manage to make any progress on this @LucaMIMOUNI ?

futnuh, 2025-11-02:
I've got the same plugin loader warning with my Tormach robot but I don't have the problem you describe - my interactive marker works as expected. I.e., I wouldn't jump to the conclusion this is the source of the problem. If possible, rerun with a higher level of debugging to see if any further details appear in the console output.

Rayten-Lo, 2026-09-18:
I reproduced the same issue and found a workaround that works reliably in my environment.

### Environment

- Ubuntu 22.04
- ROS 2 Humble
- MoveIt 2: `2.5.9`
- `moveit_ros_visualization`: `2.5.9`
- RViz 2: `11.2.26`
- Gazebo Sim
- Custom 6-DOF manipulator
- KDL kinematics plugin

### Observed behavior

The interactive marker is displayed correctly and follows the mouse while
dragging. However, after releasing the mouse, the marker returns to its
previous pose and the MotionPlanning goal state is not updated.

Joint-state goals still work correctly, and MoveIt can successfully compute
IK, plan, and execute trajectories.

### Workaround

I created a small external ROS 2 node that performs the missing final
goal-state update.

The node:

1. Subscribes to the MoveIt RViz interactive-marker feedback topic:

   ```text
   /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/feedback
   ```

2. Processes only `InteractiveMarkerFeedback.MOUSE_UP` events from
   `EE:goal_*` markers.

3. Uses the latest `/joint_states` message as the IK seed.

4. Calls MoveIt's existing `/compute_ik` service.

5. Publishes the resulting `moveit_msgs/msg/RobotState` to:

   ```text
   /rviz/moveit/update_custom_goal_state
   ```

After publishing the IK solution to this topic, the marker remains at the
released pose and the resulting goal can be planned and executed normally.

### Python implementation

```python
#!/usr/bin/env python3

import copy
import threading
import time

import rclpy
from builtin_interfaces.msg import Duration
from moveit_msgs.msg import MoveItErrorCodes, RobotState
from moveit_msgs.srv import GetPositionIK
from rclpy.node import Node
from sensor_msgs.msg import JointState
from visualization_msgs.msg import InteractiveMarkerFeedback

class RvizMarkerIkBridge(Node):
    def __init__(self):
[... 199 lines cut ...]

It does not fix the `class_loader` namespace-collision warning or identify
why the native MotionPlanning interactive-marker callback fails to retain the
goal state.

It only restores the missing final update by converting the marker pose to a
MoveIt `RobotState` through `/compute_ik`, then sending that state through the
MotionPlanning plugin's existing
`/rviz/moveit/update_custom_goal_state` interface.

The implementation currently sets:

```python
request.ik_request.avoid_collisions = False
```

Therefore the IK call itself does not reject collision states. Normal MoveIt
planning and collision checking still occur when the user presses `Plan`.

The current `/joint_states` value is used as the IK seed, so the selected IK
solution will generally be close to the robot's current configuration.

I hope this helps narrow down whether the problem is in the native
interactive-marker feedback handling or in the final query-goal-state
update.