moveit/moveit2 #3499: Rolling / Kilted package release?

urfeex, 2025-06-10:
### Description

The binary package of moveit_servo currently cannot be installed due to a missing dependency. This has already been fixed in #3474 but no binary package has been released since then.

I know, maintenance resources on Moveit open-source are currently a bit rare, but I hope that it would still be possible to do a binary release including that fix.

Thank you!

### ROS Distro

Rolling

### OS and version

Ubuntu 24.04

### Source or binary build?

Binary

### If binary, which release version?

2.13.2

### Which RMW are you using?

FastRTPS

### Steps to Reproduce

Try installing moveit_servo on a Rolling system:

```console
$ sudo apt install ros-rolling-moveit-servo                                                                                                                                                                                                                                                                              
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-rolling-moveit-servo : Depends: ros-rolling-gripper-controllers but it is not installable
E: Unable to correct problems, you have held broken packages.
```

### Expected behavior

The package should install

### Actual behavior

See reproduction section. The package doesn't install due to a missing dependency

urfeex, 2025-06-13:
I hope you don't mind me friendly poking this. Do you think, we could try to get that in before the Kilted sync coming up on Monday? It's technically a regression fix, so it might get merged still.

urfeex, 2025-06-25:
@nbbrooks I see that you got added to https://github.com/ros2-gbp/ros2-gbp-github-org/blob/latest/moveit.tf so I guess a release would be possible now?

mjforan, 2025-09-09:
Still broken on ARM (aarch64).

v4hn, 2025-09-09:
> [Failing rolling build due to unreleased patches]. I guess, another patch release for moveit2 would be great :-)

2.14.1 is out for [rolling](https://github.com/ros/rosdistro/pull/47578) (and [kilted](https://github.com/ros/rosdistro/pull/47579) though less relevant).
But [more is incoming anyway](https://github.com/moveit/moveit2/pull/3567).

> Still broken on ARM (aarch64).

Context and details missing.

---

I will close this issue to avoid keeping it open forever.
Let's stick with version-specific release requests instead of a never-ending "release NOW please".

closed 2025-09-09 by v4hn