moveit/moveit2 #3527: No candidate for `ros-rolling-moveit-ros-planning-interface` on Noble

MatthijsBurgh, 2025-07-17:
### Description

There seems to be no candidate `ros-rolling-moveit-ros-planning-interface` on Noble, while there is for `ros-rolling-moveit-ros-planning`.

### ROS Distro

Rolling

### OS and version

24.04

### Source or binary build?

Binary

### If binary, which release version?

None...

### Which RMW are you using?

None

### Steps to Reproduce

apt-cache policy ros-rolling-moveit-ros-planning-interface

### Expected behavior

`ros-rolling-moveit-ros-planning-interface` has been released

### Actual behavior

`ros-rolling-moveit-ros-planning-interface` has not been released

### Backtrace or Console output

```bash
$ apt-cache policy ros-rolling-moveit-ros-planning
ros-rolling-moveit-ros-planning:
  Installed: (none)
  Candidate: 2.14.0-1noble.20250703.130435
  Version table:
     2.14.0-1noble.20250703.130435 500
        500 http://packages.ros.org/ros2/ubuntu noble/main amd64 Packages
```
```bash
$ apt-cache policy ros-rolling-moveit-ros-planning-interface
ros-rolling-moveit-ros-planning-interface:
  Installed: (none)
  Candidate: (none)
  Version table:
```

MatthijsBurgh, 2025-07-18:
I assume these will be back with the next sync?

sea-bass, 2025-07-18:
I believe so!

MatthijsBurgh, 2025-07-18:
I am fine with closing this issue for now. Or do you want to keep it open and close it on the next sync?

MatthijsBurgh, 2025-09-07:
Should not be marked stale

MatthijsBurgh, 2025-09-22:
Sync happened https://discourse.openrobotics.org/t/new-packages-for-ros-2-rolling-ridley-2025-09-05/49929

closed 2025-09-22 by MatthijsBurgh