moveit/moveit2 #3541: Configuration Assistant exits

lukaswrnr, 2025-07-31:
### Description

Heyo, 
im using Ubuntu 22.04 and ROS2 Humble.

I generated and URDF file out of my Universal Robot ur_description xacro:
``` 
xacro ur.urdf.xacro \
  ur_type:=ur5e \
  name:=ur5e_robot \
  robot_ip:=#.#.#.# \
  use_fake_ha:=false \
  > ~/robot_ws/src/Universal_Robots_ROS2_Description/urdf/ur5e_robot.urdf
```

The ur_description repo is up to date i checked the urdf file but i do not see any issues.
I made a simple pole urdf file which loads without issues. I had the setup assistant running a few months ago with the same xacro file, i cannor really identifye the problem here. Someone knows what could work ? Thanks in advance !

### ROS Distro

Humble

### OS and version

Ubuntu 22.04

### Source or binary build?

Source

### Which RMW are you using?

None

### Steps to Reproduce

Generate the urdf file and try to load it into the setup assistant.

### Expected behavior

I can build a configuration for my roboter setup

### Actual behavior

Program closes

### Backtrace or Console output

```
[INFO] [launch]: All log files can be found below /home/onm/.ros/log/2025-07-31-12-40-08-131421-IPT-D-0337-48761
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [moveit_setup_assistant-1]: process started with pid [48774]
[moveit_setup_assistant-1] [INFO] [1753958446.444783344] [moveit_setup_assistant.urdf]: Loaded ur5e_robot robot model.
[moveit_setup_assistant-1] [INFO] [1753958446.445070168] [moveit_robot_model.robot_model]: Loading robot model 'ur5e_robot'...
[moveit_setup_assistant-1] [INFO] [1753958446.445101865] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[moveit_setup_assistant-1] [INFO] [1753958447.847507874] [moveit_setup_assistant.urdf]: Loaded ur5e_robot robot model.
[moveit_setup_assistant-1] [INFO] [1753958447.847561335] [moveit_robot_model.robot_model]: Loading robot model 'ur5e_robot'...
[moveit_setup_assistant-1] [INFO] [1753958447.847576934] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[moveit_setup_assistant-1] [INFO] [1753958448.353531668] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00168117 seconds
[moveit_setup_assistant-1] [INFO] [1753958448.353568891] [moveit_robot_model.robot_model]: Loading robot model 'ur5e_robot'...
[moveit_setup_assistant-1] [INFO] [1753958448.353576358] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[moveit_setup_assistant-1] [INFO] [1753958448.595718258] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00130879 seconds
[moveit_setup_assistant-1] [INFO] [1753958448.595740733] [moveit_robot_model.robot_model]: Loading robot model 'ur5e_robot'...
[moveit_setup_assistant-1] [INFO] [1753958448.595748438] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[moveit_setup_assistant-1] Stack trace (most recent call last):
[moveit_setup_assistant-1] #31   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4", at 0x7cd5fad762b7, in 
[moveit_setup_assistant-1] #30   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4", at 0x7cd5fad20d3a, in g_main_context_dispatch
[moveit_setup_assistant-1] #29   Object "/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5.15.3", at 0x7cd5f0563d6d, in 
[moveit_setup_assistant-1] #28   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.15.3", at 0x7cd5fc916a2b, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
[moveit_setup_assistant-1] #27   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.15.3", at 0x7cd5fc941306, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
[moveit_setup_assistant-1] #26   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7cd5fc4b9e39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
[moveit_setup_assistant-1] #25   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7cd5fd16c712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
[moveit_setup_assistant-1] #24   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7cd5fd1cbfd4, in 
[moveit_setup_assistant-1] #23   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7cd5fd1c8d3f, in 
[moveit_setup_assistant-1] #22   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7cd5fd172e46, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool)
[moveit_setup_assistant-1] #21   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7cd5fc4b9e39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
[moveit_setup_assistant-1] #20   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7cd5fd174363, in QApplication::notify(QObject*, QEvent*)
[moveit_setup_assistant-1] #19   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7cd5fd16c712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
[moveit_setup_assistant-1] #18   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7cd5fd1af4ed, in QWidget::event(QEvent*)
[moveit_setup_assistant-1] #17   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7cd5fd2641e6, in QAbstractButton::mouseReleaseEvent(QMouseEvent*)
[moveit_setup_assistant-1] #16   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7cd5fd263fc3, in 
[moveit_setup_assistant-1] #15   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7cd5fd26239d, in 
[moveit_setup_assistant-1] #14   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7cd5fd262115, in QAbstractButton::clicked(bool)
[moveit_setup_assistant-1] #13   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7cd5fc4f17c7, in 
[moveit_setup_assistant-1] #12   Source "./src/start_screen_widget.cpp", line 273, in loadFilesClick [0x7cd5e853fb89]
[moveit_setup_assistant-1] #11   Source "./src/start_screen_widget.cpp", line 414, in loadNewFiles [0x7cd5e853f8c4]
[moveit_setup_assistant-1] #10   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7cd5fc513111, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
[moveit_setup_assistant-1] #9    Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4", at 0x7cd5fad1e3e2, in g_main_context_iteration
[moveit_setup_assistant-1] #8    Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4", at 0x7cd5fad762b7, in 
[moveit_setup_assistant-1] #7    Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4", at 0x7cd5fad20d3a, in g_main_context_dispatch
[moveit_setup_assistant-1] #6    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7cd5fc513a66, in 
[moveit_setup_assistant-1] #5    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7cd5fc4bcf26, in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
[moveit_setup_assistant-1] #4    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7cd5fc4b9e39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
[moveit_setup_assistant-1] #3    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7cd5fd16c712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
[moveit_setup_assistant-1] #2    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7cd5fc4e741d, in QObject::event(QEvent*)
[moveit_setup_assistant-1] #1    Object "/opt/ros/humble/lib/librviz_common.so", at 0x7cd5fda0a8d2, in rviz_common::properties::PropertyTreeModel::propertyHiddenChanged(rviz_common::properties::Property const*)
[moveit_setup_assistant-1] #0    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7cd5fc4f128e, in 
[moveit_setup_assistant-1] Segmentation fault (Address not mapped to object [0x10000000a])
[ERROR] [moveit_setup_assistant-1]: process has died [pid 48774, exit code -11, cmd '/opt/ros/humble/lib/moveit_setup_assistant/moveit_setup_assistant --ros-args'].

```

jimmy-mcelwain, 2025-08-01:
Okay. I just downloaded the moveit2 source files and built in debug mode. I was having a bit of trouble with installing debug symbols (my fault). I am building from 3f2d7f51a31dcc03e76735684b90a9de13249cfa rather than 2.5.9, but since I'm getting the same problem I am guessing that doesn't really matter. 

It always seems to fail here: https://github.com/moveit/moveit2/blob/3f2d7f51a31dcc03e76735684b90a9de13249cfa/moveit_setup_assistant/moveit_setup_core_plugins/src/start_screen_widget.cpp#L414

I have attached the full stack trace below, but the part that I think is important is:

```
#1  0x00007ffff7b048d3 in rviz_common::properties::PropertyTreeModel::propertyHiddenChanged(rviz_common::properties::Property const*) (this=<optimized out>, _t1=<optimized out>)
    at ./.obj-x86_64-linux-gnu/include/rviz_common/properties/moc_property_tree_model.cpp:171
        _a = {0x0, 0x7fffffff9f58}
```

On my working PC, I have rviz 11.2.16. On my PC with the problem, I have rviz 11.2.19. I think that something in an rviz update between those two introduced some sort of incompatibility. 

I removed my rviz packages, built 11.2.19 from source, and ran into the same problem. After downgrading to 11.2.16 (and later on 11.2.18) and building from source, I am no longer running into the problem. 

I'm unfamiliar with these codebases, and I don't know whether the problem lies with rviz or moveit2, though. But for me, downgrading to 11.2.16 or 11.2.18 worked. So I'm guessing that something in rviz 11.2.19 broke compatibility. https://github.com/ros2/rviz/compare/11.2.18...ros2:11.2.19

Here is the new stack trace on seg fault with moveit2 2.5.9 and rviz 11.2.19:

```
bizon@bizon-desktop:~/workspaces/moveit$ QT_DEBUG_PLUGINS=1 ros2 run --prefix 'gdb --args' moveit_setup_assistant moveit_setup_assistant 
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04.2) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/bizon/workspaces/moveit/install/moveit_setup_assistant/lib/moveit_setup_assistant/moveit_setup_assistant...
(gdb) run
Starting program: /home/bizon/workspaces/moveit/install/moveit_setup_assistant/lib/moveit_setup_assistant/moveit_setup_assistant 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffec9fe640 (LWP 83083)]
[New Thread 0x7fffec1fd640 (LWP 83084)]
[New Thread 0x7fffeb975640 (LWP 83085)]
[New Thread 0x7fffeb174640 (LWP 83086)]
[New Thread 0x7fffea973640 (LWP 83087)]
[New Thread 0x7fffea172640 (LWP 83088)]
[New Thread 0x7fffe9971640 (LWP 83089)]
[New Thread 0x7fffe90b0640 (LWP 83090)]
[New Thread 0x7fffe88af640 (LWP 83091)]
QFactoryLoader::QFactoryLoader() checking directory path "/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms" ...
QFactoryLoader::QFactoryLoader() looking at "/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqeglfs.so"
Found metadata in lib /usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqeglfs.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "eglfs"
        ]
    },
    "archreq": 0,
    "className": "QEglFSIntegrationPlugin",
    "debug": false,
    "version": 331520
}

Got keys from plugin meta data ("eglfs")
QFactoryLoader::QFactoryLoader() looking at "/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqlinuxfb.so"
Found metadata in lib /usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqlinuxfb.so, metadata=
[... 449 lines cut ...]
#28 0x00007ffff6941307 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#29 0x00007ffff6916a2c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#30 0x00007fffe38f8d6e in  () at /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#31 0x00007ffff3b20d3b in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#32 0x00007ffff3b762b8 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#33 0x00007ffff3b1e3e3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#34 0x00007ffff65130b8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#35 0x00007ffff64b875b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#36 0x00007ffff64c0cf4 in QCoreApplication::exec() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#37 0x00005555555e904a in main(int, char**) (argc=1, argv=0x7fffffffbda8) at /home/bizon/workspaces/moveit/src/moveit2/moveit_setup_assistant/moveit_setup_assistant/src/main.cpp:112
[... cut ...] ount 1, weak count 0) = {get() = 0x5555563de670}, std::shared_ptr<moveit_setup::SetupStepWidget> (use count 1, weak count 0) = {get() = 0x555556329890}, std::shared_ptr<moveit_setup::SetupStepWidget> (use count 1, weak count 0) = {get() = 0x5555563e1fe0}, std::shared_ptr<moveit_setup::SetupStepWidget> (use count 1, weak count 0) = {get() = 0x5555563de0f0}}, config_data_ = std::shared_ptr<moveit_setup::DataWarehouse> (use count 26, weak count 1) = {get() = 0x555555be1a10}}
        result = 0

```

CanavarB, 2025-08-21:
how to downgrade?

jimmy-mcelwain, 2025-08-21:
@CanavarB I would recommend building from source. Clone https://github.com/ros2/rviz and check out `11.2.18` (each version is tagged), then build it (see [here](https://github.com/ros2/rviz?tab=readme-ov-file#building-rviz-in-a-separate-workspace)). Make sure to source the overlay workspace after building and then launch the MoveIt Setup Assistant in the terminal with `rviz2` `11.2.18` sourced.

CanavarB, 2025-08-24:
Okay. Unlike the know-it-alls here, for those experiencing this issue, I'll explain step-by-step how to solve it. First, the problem is with the visualization part of Setup Assistant. Setup Assistant uses RViz to visualize URDFs. Since our clever friends made incompatible updates without testing it (even in the LTS version :D)  and didn't even keep the older versions in the ROS repo, we need to recompile the older version of RViz and provide it to Setup Assistant ourselves. So, which version should we downgrade to? First, we need to find the date when Setup Assistant started having problems from our most recent upgrades and install the version when it wasn't causing any problems.

You can find the update logs here:

```
/var/log/apt/history.log
```

For example, I encountered this issue because I upgraded from 14.1.12 to 14.1.13. **Yours may be different.**
After finding the version, remove all rviz-related packages on the system (the real problem is the rviz-common package).

### Note: this process removes almost everything related to ROS, but it's necessary.

``` bash
sudo apt remove ros-<YOUR_ROS_VERSION_CODE>-rviz* --purge
```

Clone the RViz GitHub repo and set it to your desired version.

``` bash
mkdir -p ~/rviz2_ws/src
cd ~/rviz2_ws/src
git clone https://github.com/ros2/rviz.git
cd rviz
git checkout <YOUR_RVIZ_VERSION>
```
You can check tags name with:

```bash
git tag
```

Then compile:

``` bash
cd ..
colcon build --merge-install
```

Whenever you want to use the Moveit Setup Assistant, source this workspace first. The Moveit Setup Assistant will use this version of RViz.

agrotrushant, 2025-08-26:
Thanks for the process, I had the similar issue and was able to fix it.

wildper, 2025-09-13:
thanks, l thougth of many possibilities,but l didn't expect the official installation to have conflicts

Ghislain04, 2025-11-05:
I am having the same issue but i am not able to fix it. I downgraded rviz but it is not working. I tested with Humble and Jazzy but i have still no solution

jimmy-mcelwain, 2025-11-05:
@Ghislain04 You're sure that you are running the correct version of `rviz2`? If you type `ros2 pkg xml rviz2`, what shows in the `version` tag? Does your error log match what is described in the comments above? 

According to https://github.com/moveit/moveit2/issues/3576#issuecomment-3445808718, https://github.com/moveit/moveit2/pull/3604 fixes this issue. I have not investigated that at all though and I cannot confirm that it is a solution.

Ghislain04, 2025-11-07:
Okay thank you that's finaly working. I might have had forgoten sourcing or something 

For Jazzy i used:

1. `wget http://snapshots.ros.org/humble/2025-06-18/ubuntu/pool/main/r/ros-jazzy-rviz-common_14.1.11-1noble.20250520.201719_amd64.deb`
2. `sudo dpkg -i ros-jazzy-rviz-common_14.1.11-1noble.20250520.201719_amd64.deb`
3. `source install/setup.bash`

For Humble, it's working too.