moveit/moveit2 #3559: moveit_servo with STL collision mesh causes infinitesimal changes

christian-rauch, 2025-08-20:
### Description

I am using a Franka description with a wrist-mounted camera. The camera collision model is an STL and is connected via another link, again with STL collision mesh. For this configuration, `servo_keyboard_input` produces tiny joint position updates. If I remove the STL collision mesh, I get larger joint position updates as expected.

### ROS Distro

Jazzy

### OS and version

Ubuntu 24.04

### Source or binary build?

Binary

### If binary, which release version?

2.12

### Which RMW are you using?

FastRTPS

### Steps to Reproduce

1. install packages from https://github.com/mul-cps/mul_franka and their dependencies
2. run the fake controllers `ros2 launch mul_franka_launch franka_moveit_camera.launch.py fake:=true camera:="femto"`
3. run servo_keyboard_input: `ros2 run moveit_servo servo_keyboard_input`
4. `ros2 topic echo /panda_arm_controller/joint_trajectory`
5. select twist and press forward, observe that the second joint changes only very slowly, e.g. by 0.0001 per second

If I remove the>
```XML
<collision>
  <geometry>
    <mesh filename="package://mul_franka_description/meshes/visual/Femto_Bolt_Camera_Mount.stl" scale="0.001 0.001 0.001"/>
  </geometry>
</collision>
```
from file `panda_femto_bolt_camera_mount.xacro`, the servoing behaves as expected.

### Expected behavior

Servoing should work with collision meshes.

### Actual behavior

The changes are too small to notice.

AndyZe, 2025-08-20:
The camera link is probably just close to collision with another link, thus Servo allows down. That's the default behavior. Consider disabling collision checking between the camera link and that other link in the SRDF.

christian-rauch, 2025-08-20:
I already defined a couple of `<disable_collisions link1="A" link2="B" reason="Adjacent"/>` between the camera, mount and robot links in the SRDF. I am using the same URDF and SRDF files with regular MoveIt without issues. Therefore, I assumed that collisions should not be an issue. Do I have to specifically configure moveit_servo differently?

AndyZe, 2025-08-20:
Nice. Probably what was going on is, there is about 1mm of distance between the mount and the camera stl, so the binary collision check didn't trigger. That's why normal MoveIt motion planning succeeded.

closed 2025-08-20 by AndyZe