moveit/moveit2 #3565: TOTG paths can contain extremely high accelerations

Eruvae, 2025-09-02:
### Description

The time optimal trajectory generation implementation sometimes returns paths that contain extremely high accelerations. I tried to track down the issue by placing asserts and additional logs in the implementation, and the underlying cause seems to be that the path sometimes contains two trajectory segments with almost identical time stamps, causing the time_step in the getAcceleration method [here](https://github.com/moveit/moveit2/blob/3e526af72eab5fb45d699a36736317c6e4ae0c03/moveit_core/trajectory_processing/src/time_optimal_trajectory_generation.cpp#L904) to be close to 0. Since the acceleration is divided by time_step, it is in turn extremely high. I put an example log output with logs in getAcceleration and doTimeParametrizationCalculation in the field below.

I don't know what the underlying cause for generating two trajectory segments with similar time steps is. However, to prevent this from happening, I think the best solution would be to not just use the previous waypoint to compute the acceleration, but instead sample points with getTrajectorySegment at times +- dt around the passed time, to guarantee a sufficient time_step. The current way is anyway problematic, since it is never checked whether the iterator returned by getTrajectorySegment is at the beginning of the trajectory, in which case previous would be invalid. This currently seems to never be the case, but only because getTrajectorySegment [here](https://github.com/moveit/moveit2/blob/3e526af72eab5fb45d699a36736317c6e4ae0c03/moveit_core/trajectory_processing/src/time_optimal_trajectory_generation.cpp#L855) checks for time >= cached_trajectory_segment_->time_, so even if you pass 0, you don't get the starting segment at time 0 (you would have to pass a negative time).

Aside from this, while debugging, I also noticed some inconsistent use of joint indices in the computeTimeStamps methods. [Here](https://github.com/moveit/moveit2/blob/3e526af72eab5fb45d699a36736317c6e4ae0c03/moveit_core/trajectory_processing/src/time_optimal_trajectory_generation.cpp#L957), vars[active_joint_indices[idx]] is used to retrieve the joint name (which should be correct), but [here](https://github.com/moveit/moveit2/blob/3e526af72eab5fb45d699a36736317c6e4ae0c03/moveit_core/trajectory_processing/src/time_optimal_trajectory_generation.cpp#L965), [here](https://github.com/moveit/moveit2/blob/3e526af72eab5fb45d699a36736317c6e4ae0c03/moveit_core/trajectory_processing/src/time_optimal_trajectory_generation.cpp#L973) and [here](https://github.com/moveit/moveit2/blob/3e526af72eab5fb45d699a36736317c6e4ae0c03/moveit_core/trajectory_processing/src/time_optimal_trajectory_generation.cpp#L986), vars[idx] is used, which should be wrong. On the other hand, the function [here](https://github.com/moveit/moveit2/blob/3e526af72eab5fb45d699a36736317c6e4ae0c03/moveit_core/trajectory_processing/src/time_optimal_trajectory_generation.cpp#L1062) uses a different loop, going directly through the values of the indices vector. But that also means that e.g. when setting max_velocity [here](https://github.com/moveit/moveit2/blob/3e526af72eab5fb45d699a36736317c6e4ae0c03/moveit_core/trajectory_processing/src/time_optimal_trajectory_generation.cpp#L1072), this could be in a different order than the max_velocity [here](https://github.com/moveit/moveit2/blob/3e526af72eab5fb45d699a36736317c6e4ae0c03/moveit_core/trajectory_processing/src/time_optimal_trajectory_generation.cpp#L968), since idx is not the same.

I'll try to create a PR addressing these issues.

### ROS Distro

Humble

### OS and version

Ubuntu 22.04

### Source or binary build?

Source

### If source, which branch?

main

### Which RMW are you using?

None

### Steps to Reproduce

Repeatedly compute trajectories using TOTG, check if acceleration values are within limits.

### Expected behavior

Acceleration should always stay within limits.

### Actual behavior

In rare instances, TOTG can return extremely high accelerations exceeding the robot limits.

### Backtrace or Console output

```
sampling at time: 0, duration: 4.39519
prev time: 0, it time: 0.001
time_step: 0.001, acceleration: 0.817481, path_pos: 4.08741e-07, path_vel: 0.000817481, path_acc: -0.260785  0.264768  0.337992 0.0853275 -0.231163 -0.174097      0.57
position:  0.683  0.598 -1.797 -0.793 -0.264  1.183 -0.859, velocity: -0.000260785  0.000264768  0.000337992  8.53275e-05 -0.000231163 -0.000174097      0.00057, acceleration: -0.260785  0.264768  0.337992 0.0853275 -0.231163 -0.174097      0.57
sampling at time: 0.05, duration: 4.39519
prev time: 0.049, it time: 0.05
time_step: 0.001, acceleration: 0.817481, path_pos: 0.00102185, path_vel: 0.0408741, path_acc: -0.260785  0.264768  0.337992 0.0853275 -0.231163 -0.174097      0.57
position:  0.682674  0.598331  -1.79658 -0.792893 -0.264289   1.18278 -0.858288, velocity:  -0.0130392   0.0132384   0.0168996  0.00426637  -0.0115581 -0.00870484      0.0285, acceleration: -0.260785  0.264768  0.337992 0.0853275 -0.231163 -0.174097      0.57
sampling at time: 0.1, duration: 4.39519
prev time: 0.099, it time: 0.1
time_step: 0.001, acceleration: 0.817481, path_pos: 0.00408741, path_vel: 0.0817481, path_acc: -0.260785  0.264768  0.337992 0.0853275 -0.231163 -0.174097      0.57
position:  0.681696  0.599324  -1.79531 -0.792573 -0.265156   1.18213  -0.85615, velocity: -0.0260785  0.0264768  0.0337992 0.00853275 -0.0231163 -0.0174097      0.057, acceleration: -0.260785  0.264768  0.337992 0.0853275 -0.231163 -0.174097      0.57
sampling at time: 0.15, duration: 4.39519
prev time: 0.149, it time: 0.15
time_step: 0.001, acceleration: 0.817481, path_pos: 0.00919667, path_vel: 0.122622, path_acc: -0.260785  0.264768  0.337992 0.0853275 -0.231163 -0.174097      0.57
position:  0.680066  0.600979   -1.7932  -0.79204 -0.266601   1.18104 -0.852588, velocity: -0.0391177  0.0397152  0.0506987  0.0127991 -0.0346744 -0.0261145     0.0855, acceleration: -0.260785  0.264768  0.337992 0.0853275 -0.231163 -0.174097      0.57
sampling at time: 0.2, duration: 4.39519
prev time: 0.199, it time: 0.201
time_step: 0.002, acceleration: 0.817481, path_pos: 0.0165135, path_vel: 0.164314, path_acc: -0.260785  0.264768  0.337992 0.0853275 -0.231163 -0.174097      0.57
position:  0.677784  0.603295  -1.79024 -0.791293 -0.268623   1.17952   -0.8476, velocity: -0.0524177  0.0532184  0.0679363  0.0171508 -0.0464637 -0.0349935    0.11457, acceleration: -0.260785  0.264768  0.337992 0.0853275 -0.231163 -0.174097      0.57
sampling at time: 0.25, duration: 4.39519
prev time: 0.249, it time: 0.25
time_step: 0.001, acceleration: 0.817481, path_pos: 0.0255463, path_vel: 0.20437, path_acc: -0.260785  0.264768  0.337992 0.0853275 -0.231163 -0.174097      0.57
position:   0.67485  0.606274  -1.78644 -0.790334 -0.271224   1.17756 -0.841188, velocity: -0.0651962  0.0661921  0.0844979  0.0213319 -0.0577907 -0.0435242     0.1425, acceleration: -0.260785  0.264768  0.337992 0.0853275 -0.231163 -0.174097      0.57
sampling at time: 0.3, duration: 4.39519
[... 76 lines cut ...]
prev time: 1.249, it time: 1.25
time_step: 0.001, acceleration: 0.817481, path_pos: 0.638657, path_vel: 1.02185, path_acc: -0.260785  0.264768  0.337992 0.0853275 -0.231163 -0.174097      0.57
position:  0.479262   0.80485  -1.53294 -0.726338 -0.444596   1.04699 -0.413688, velocity: -0.325981   0.33096  0.422489  0.106659 -0.288953 -0.217621    0.7125, acceleration: -0.260785  0.264768  0.337992 0.0853275 -0.231163 -0.174097      0.57
sampling at time: 1.3, duration: 4.39519
prev time: 1.299, it time: 1.3
time_step: 0.001, acceleration: 0.817481, path_pos: 0.690772, path_vel: 1.06273, path_acc: -0.260785  0.264768  0.337992 0.0853275 -0.231163 -0.174097      0.57
position:  0.462637  0.821729   -1.5114 -0.720898 -0.459332   1.03589  -0.37735, velocity:  -0.33902  0.344199  0.439389  0.110926 -0.300511 -0.226326     0.741, acceleration: -0.260785  0.264768  0.337992 0.0853275 -0.231163 -0.174097      0.57
sampling at time: 1.35, duration: 4.39519
prev time: 1.349, it time: 1.35
time_step: 0.001, acceleration: 0.817481, path_pos: 0.74493, path_vel: 1.1036, path_acc: -0.260785  0.264768  0.337992 0.0853275 -0.231163 -0.174097      0.57
position:   0.44536   0.83927  -1.48901 -0.715245 -0.474647   1.02435 -0.339588, velocity: -0.352059  0.357437  0.456289  0.115192  -0.31207 -0.235031    0.7695, acceleration: -0.260785  0.264768  0.337992 0.0853275 -0.231163 -0.174097      0.57
[... cut ...]
time_step: 2.22045e-16, acceleration: -2.03763e+15, path_pos: 0.919667, path_vel: 0.773778, path_acc:  6.50023e+14 -6.59953e+14 -8.42467e+14 -2.12685e+14  5.76189e+14  4.33948e+14 -1.42076e+15
position:  0.389617  0.895864  -1.41676 -0.697007 -0.524058  0.987141  -0.21775, velocity: -0.246843  0.250613  0.319922 0.0807658 -0.218804 -0.164789  0.539527, acceleration:  6.50023e+14 -6.59953e+14 -8.42467e+14 -2.12685e+14  5.76189e+14  4.33948e+14 -1.42076e+15
Assertion `std::fabs(acceleration[j]) <= max_acceleration[j] * 2' failed.
```

ahmedsleem109, 2026-11-07:
@Eruvae are you still planning to open a PR for this? If not, I'd like to pick it up and build on your reproduction test (with credit, of course).

Here is what I think happens. When `integrateBackward()` splices the backward curve into the forward one, the first step of the backward part can end up almost at the same `path_pos_` as the last kept forward step. The timing loop at the end of `Trajectory::create()` then gives those two steps a time gap of about 1e-16, and `getAcceleration()` divides by it.

Plan:
1. Add your hexfloat input as a regression test that checks all sampled accelerations stay within limits.
2. Fix it at the source: while computing time stamps, drop steps whose time gap to the previous step is below a small epsilon, so the trajectory never has near-zero segments.
3. Also make `getPosition()` / `getVelocity()` / `getAcceleration()` safe for a zero `time_step`, and handle the case where `getTrajectorySegment()` returns the first step.

I'd keep the joint index mismatch in `computeTimeStamps()` for a separate PR, so each change stays small and easy to review.

Eruvae, 2026-11-07:
@ahmedsleem109 Go ahead, I'm not working on this anymore. At the time, I tried to add an acceleration variable into the integration process to completely avoid the divide by time_step, but this caused other tests to fail, so I wasn't sure if it was the right approach. I then opted for a local workaround, similar to what you mention in 2), where we just loop through the points at the end of Trajectory::create and remove any with time diff < epsilon. But while that worked to avoid infinite acceleration, I didn't think it was a proper solution, so I didn't submit a PR for that. It also doesn't guarantee that the specified maximum acceleration is maintained, I have observed that acceleration limits are exceeded.

ahmedsleem109, 2026-11-07:
Thanks @Eruvae, that helps a lot. Since dropping the tiny steps still leaves accelerations over the limit, I'll look for where the velocity jump at the join comes from, and make the test check the actual acceleration limits instead of only catching huge values.